VUnit/vunit #257: Failure on adding external libraries in the VUnit project from the installed modelsim.ini

gsorrenti, 2017-06-13:
Hello everybody I experimented the following Modelsim error running vunit/examples/vhdl/uart test updating from commit f163bde 21/02/2017 to head 358493e 12/05/2017.
```
Starting tb_uart_lib.tb_uart_tx.test_send_many_bytes
# vsim -modelsimini f:/github/vunit/vunit_out/modelsim/modelsim.ini -wlf f:/github/vunit/vunit_out/test_output/324f957a68a62f3323566c903f238d45a63b1080/modelsim/modelsim/vsim.wlf -quiet -t ps
-onfinish stop tb_uart_lib.tb_uart_tx(tb) -L vunit_lib -L osvvm -L uart_lib -L tb_uart_lib -L mtipa -L floatfixlib -L mtiovm -L sv_std -L ieee_env -L ieee -L mtiuvm -L mc2_lib -L vital2000 -L
verilog -L mtiupf -L synopsys -L mtiavm -L modelsim_lib -L vhdlopt_lib -L infact -L std -L std_developerskit -L mtirnm -L mgc_ams -g/tb_uart_tx/runner_cfg="active python runner : true,enabled_test_cases : test_send_many_bytes,output path : f::/github/vunit/vunit_out/test_output/324f957a68a62f3323566c903f238d45a63b1080/,tb path : f::/github/vunit/examples/vhdl/uart/src/test/"
# Start time: 11:52:42 on May 29,2017
# ** Fatal: (vsim-7019) Can't locate a C/C++ compiler for 'DPI Export Compilation'.
#
#
# FATAL ERROR while loading design
# Error loading design
Error loading design
# End time: 11:52:42 on May 29,2017, Elapsed time: 0:00:00
# Errors: 1, Warnings: 0
```
The failure was introduced in 358493e. The error disappear in commit 358493e commenting lines 112-121 in modelsim_interface.py, but the error exists if I move those lines above line 136

kraigher, 2017-06-17:
Have a look a c4c2035 and see if it works better.

gsorrenti, 2017-06-19:
It works. Thanks.

closed 2017-06-19 by gsorrenti