VUnit/vunit #260: VUNit + Windows + Cygwin + Aldec RPRO

svenka3, 2017-07-11:
Hello,
  I am facing some strange issues with VUnit. Below is the setup:

--> Windows (64b) + Cygwin
--> Aldec Riviera-PRO (RPRO)
--> VUnit Verilog/SV (user_guide example)

Under Cygwin I am able to manually compile VUnit (Though at run time I get  	 $error("Internal error: Cannot find 'enabled_test_cases' key"); - will file a separate issue later if needed).

But when I run "python run.py" (user_guide example) I get this error on "no simulator in PATH". It is strange because I can run "blog/vsim" on command line. I also tried setting VUNIT_SIMULATOR with no luck. 

echo $PATH also shows the path to RPRO properly.

Any clues please?

Thanks
Srini

kraigher, 2017-07-11:
Try the commit referenced above first and let me know if it works.
Otherwise if the automatic toolchain detection does not work you can always set the `VUNIT_RIVIERAPRO_PATH` environment variable to the folder containing the vsim binaries such as `/c/Aldec/Riviera-PRO-2016.06/bin/`

svenka3, 2017-07-11:
Thanks, that helped a bit more. Now I am stuck at classical Windows vs. Cygwin path issues. I believe the answer is around: 

https://stackoverflow.com/questions/7752052/how-to-switch-from-posix-to-windows-with-cygwin?rq=1

Where exactly am I supposed to fix that? In simulator_interface.py? 

The error I see now is:

Compiling ../../../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.0.1rc0-py2.7.egg/vunit/verilog/vunit_pkg.sv into vunit_lib ...
ALOG: Error: Cannot initialize ULM with given configuration file `/home/srini/proj/srini/VW_VUnit/vunit/examples/verilog/user_guide/vunit_out/rivierapro/library.cfg'.
Failed to compile ../../../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.0.1rc0-py2.7.egg/vunit/verilog/vunit_pkg.sv with command:
/cygdrive/c/tools/EDA/Aldec/Riviera-PRO-2016.06-x64/bin/vlog -quiet -sv2k12 -lc /home/srini/proj/srini/VW_VUnit/vunit/examples/verilog/user_guide/vunit_out/rivierapro/library.cfg -work vunit_lib /usr/lib/python2.7/site-packages/vunit_hdl-2.0.1rc0-py2.7.egg/vunit/verilog/vunit_pkg.sv -l vunit_lib -l lib +incdir+/usr/lib/python2.7/site-packages/vunit_hdl-2.0.1rc0-py2.7.egg/vunit/verilog/include

kraigher, 2017-07-11:
If cygwin generates paths which other Windows programs cannot understand it will be hard to fix. Maybe place the vunit_out folder and source files somewhere that normal Windows can understand?

svenka3, 2017-07-11:
I'm starting to hate this combination - Windows + Cygwin for this trial. I'm a big fan of Linux and hence resort to Cygwin on any Windows machine. Maybe I shouldn't..

 I am sure I need more patience, but for now it is a free trial and have limited bandwidth for this experiment. Perhaps we can leave this open and revisit if there are more such users. 

FWIW - I was able to play around rivierapro_interface.py and get through that library.cfg issue. Clearly it is path related, but not the "/" vs "\", rather missing absolute path (Such as C:/cygwin64/home/srini/proj vs. C:/proj" ).

Below is a code snippet that I modified and managed to get it running. Not elegant, not portable etc. Now it has problems with other source files (Of-course). 

I'm putting it on hold for now maybe.

Warm Regards
Srini 

  def compile_verilog_file_command(self, source_file):
        """
        Returns the command to compile a Verilog file
        """
	#VW
	print (self._sim_cfg_file_name)
	vw_modified_cygpath = "C:/cygwin64/home/srini" + self._sim_cfg_file_name[11:]
	print (vw_modified_cygpath)
        #vw_modified_cygpath = "C:/cygwin64/home/srini/proj/srini/VW_VUnit/vunit/examples/verilog/user_guide/vunit_out/rivierapro/library.cfg"

        args = [join(self._prefix, 'vlog'), '-quiet', '-sv2k12', '-lc', vw_modified_cygpath]

kraigher, 2017-07-11:
On Windows I tend to use git bash + [cmder](http://cmder.net/). Cygwin has never been attractive to me, just seems like working against the system.

svenka3, 2017-07-11:
Voila! I believe I have a fix that should work here.. It is not perfect (Needs if.. around those fixes to keep it Cygwin + Windows + Linux compatible), but am getting there..

Here is a handy def for this:

def vw_fix_cygwin_path(in_path):
        """
        Trick to fix dirty path issues with Cygwin
        """
	#VW
	print ("VW: " + in_path)
	# vw_modified_cygpath = "C:/cygwin64/home/srini" + self._sim_cfg_file_name[11:]
	vw_path_cmd = "cygpath -w " + in_path
	vw_modified_cygpath = subprocess.check_output(vw_path_cmd, shell=True)
	print ("VW: " + vw_modified_cygpath)
        vw_modified_cygpath = vw_modified_cygpath.replace('\\', '/')
	print ("VW: " + vw_modified_cygpath)
        vw_modified_cygpath = vw_modified_cygpath.rstrip()
	print ("VW: " + vw_modified_cygpath)

        #vw_modified_cygpath = "C:/cygwin64/home/srini/proj/srini/VW_VUnit/vunit/examples/verilog/user_guide/vunit_out/rivierapro/library.cfg"
	print ("VW: " + vw_modified_cygpath)
	return vw_modified_cygpath

[image]

closed 2017-12-20 by kraigher