VUnit/vunit #262: Modelsim - mem load command

Piressss, 2017-07-18:
I'm trying to use the "mem load command" from Modelsim, like the following example:

ui = VUnit.from_argv()
ui.set_sim_option('modelsim.vsim_flags',['../../report_top.do'])

The contents of the report_top.do file are:
mem load -i ../../hist.mem -startaddress 0 -endaddress 3145727 /m60_report_top_tb/ddr_ctrl_model_u/mig_bhv_model_u/ram_s;

But the Modelsim log this warning:

Warning: Ignoring nested -do argument: "do ../../report_top.do"

I would like to know if is it possible to load memory contents?

kraigher, 2017-07-18:
I will another `sim_option` to allow running any user defined `.do` / `.tcl` file after `vunit_load`.. I will also make a `tb_path` variable available when running the user defined file. The `tb_path` variable will contain the path to the folder containing the test bench. The script can use this variable to reference other files with a location know relative to the test bench. Would this work for you?

Piressss, 2017-07-18:
Sure, if I get your idea probably this workaround will help me.

Piressss, 2017-07-18:
Kraigher, I would like to know if your propose also will work in background or just in GUI mode?

kraigher, 2017-07-18:
I will add a `modelsim.init_files.after_load` which is a list of files that are *always* fun after loading the top level in `vunit_load`. It is almost done, I will push a commit soon for you to test.

Piressss, 2017-07-18:
Sounds good, just to you know, I used the simulation option 'modelsim.init_file.gui':

ui.set_sim_option('modelsim.init_file','modelsim.tcl')

And the TCL file contentes are:

echo "Running DO File"
do ../../report_top.do

In the GUI mode this workaround ran and the command load mem worked.

I think that your idea just make the same way but in the background.

kraigher, 2017-07-18:
Yes my idea is the same as `init_file.gui` which we will keep since. The use case for that one is waveform configuration.

kraigher, 2017-07-18:
I made a commit to the `issue_262` branch where I added `modelsim.init_files.after_load`. When running the init file the `vunit_tb_path` variable is available as the path of the folder containing the test bench.

kraigher, 2017-07-18:
Let me know if it works for you and I will push it to `master` and make a release.

Piressss, 2017-07-18:
I tried that way:

ui.set_sim_option('modelsim.init_files.after_load','report_top.do')

But something is wrong, look the result:

Sourcing file tb_path/r from modelsim.init_files.after_load
Sourcing tb_path/r failed
couldn't read file "tb_path/r": no such file or directory

I think the string (file name) was not identified.

kraigher, 2017-07-18:
Must be a list of strings

kraigher, 2017-07-18:
Thus the Plural in the name. Maybe could use some input validation

kraigher, 2017-07-19:
I added input validation of all sim and compile options now. Did you change to a list of string?

Piressss, 2017-07-19:
I changed to a list of string and it works fine.

Thanks.

kraigher, 2017-07-19:
Now part of 2.1.0 release.

closed 2017-07-19 by kraigher