VUnit/vunit #264: could not decod file test.vhd using encoding utf-8, ignoring encoding errors

daedadu, 2017-07-19:
I get this on a file I modified with emacs on Windows and python2.7
Do I get further problems because of that?
Is this a python related issue since python2.7 is a little messy with unicode or is this a vunit problem?

kraigher, 2017-07-19:
What type of file is it? VUnit should read all HDL files using latin-1 encoding.

daedadu, 2017-07-19:
The file has utf-8 encoding, like the warning correctly says

kraigher, 2017-07-19:
Both the VHDL and Verilog standard says HDL files should be encoded in latin-1 format so thus VUnit opens all HDL files in latin-1. If it cannot decode the file using latin-1 it falls back to utf-8 and issues the warning. It will most likely not affect anything since often the violating characters are found in the comments where people write their names using non-english characters etc.

closed 2017-12-22 by kraigher