VUnit/vunit #265: Copy VHDL files to "...\vunit_out\preprocessed" directory

mohsenblueice, 2017-07-20:
Hi everybody,
I am newcomer is VUNIT so I do not know my question is relevant or not.
when I run "run.py" all VHDL files from my working directory are copied into "...\vunit_out\preprocessed" directory. OK, fair enough. but the point is that if I open Modelsim using VUNIT and I need to modify the code in my working directory it does not affect and I have to close everything and rerun "run.py" ! I mean recompile in Modelsim does not affect anymore unless I go to the "...\vunit_out\preprocessed" directory and modify files there as well. 
It is strange, my friend never saw this behavior and he believes that copying files to "...\vunit_out\preprocessed" directory is not done on his PC.
is there any hint?

Mohsen

kraigher, 2017-07-20:
This is only the case when you run with preprocesing enabled which is not the default and only and experimental feature. I would not recommend using that feature if this bothers you. Even using preprocessing you can do `vunit_restart` in the gui to re-compile which will also update the preprocesing.

mohsenblueice, 2017-07-20:
Thank you Kraigher. 
I would like close the issue. Do I need to score somewhere?

kraigher, 2017-07-20:
Just press close button

closed 2017-07-20 by mohsenblueice