VUnit/vunit #267: Changes in path handling

crdavis12, 2017-07-25:
My environment is set up with the simulator (Active-HDL) not included in my system path. Instead, I have it added in the Python script that calls VUnit. As of VUnit 2.0.1, the following worked:

```
import os
from vunit import VUnit
os.environ['PATH'] = 'T:/Aldec/Active-HDL-10.3-x86/BIN'
ui = VUnit.from_argv()
ui.main()
```

Starting with VUnit 2.1.0 (confirmed also with 2.1.1), the following error occurs:
```
Traceback (most recent call last):
  File "C:\virtualenv\sff\lib\site-packages\vunit\ui.py", line 699, in main
    all_ok = self._main()
  File "C:\virtualenv\sff\lib\site-packages\vunit\ui.py", line 740, in _main
    return self._main_run()
  File "C:\virtualenv\sff\lib\site-packages\vunit\ui.py", line 749, in _main_run
    simulator_if = self._create_simulator_if()
  File "C:\virtualenv\sff\lib\site-packages\vunit\ui.py", line 743, in _create_simulator_if
    return self._simulator_factory.create(self._args, self._simulator_output_path)
  File "C:\virtualenv\sff\lib\site-packages\vunit\simulator_factory.py", line 212, in create
    raise RuntimeError("No available simulator detected. "
RuntimeError: No available simulator detected. Simulator executables must be available in PATH environment variable.
```

If I change the order a little bit, I am able to get it to work.
```
import os
os.environ['PATH'] = 'T:/Aldec/Active-HDL-10.3-x86/BIN'
from vunit import VUnit
ui = VUnit.from_argv()
ui.main()
```

It looks like the path is now being cached when vunit is imported. Is this expected behavior?

kraigher, 2017-07-25:
Yes this is the expected behavior. Environment variables controlling the simulator selection are evaluated during import.

kraigher, 2017-07-25:
Is this a problem for you? If not maybe we can close the issue?

crdavis12, 2017-07-25:
Thanks for the response. You can close it. It may be helpful to have it documented somewhere that the path must be set before import, but no reason to change the behavior.

closed 2017-07-25 by crdavis12