VUnit/vunit #269: compile.do converter

bundi78, 2017-08-03:
Does there exists any tool/scripts which converts a given ModelSim/QuetaSim compile.do file into an run.py? I have some quite large Xilinx Vivado generated files, with vhdl and verilog (incl. IncludeDirs), which i would use as starting point.

kraigher, 2017-08-04:
I do not have any tool which converts a compile.do file into appropriate VUnit run.py. It should be quite easy to do using a regex-based Python script or an editor which supports regex-based search-and-replace.

Doing this conversion migth be a quick way to get started but long term it might not be an ideal flow using Vivado. Using Vivado it is possible to automatically via TCL generate simulation libraries if all ip in a project and create a compile order file which can easily be parsed. This can be used to get automated reproducability of all artifacts and avoid checking them in to version control. We have an example of this in our repo in examples/vhdl/vivado. This example is a bit old and does not representative the best way to achieve this as of know but it could be used as inspiration. I know at least two companies who use a similar but improved approach to automate Vivado simulation library compilation and integration with VUnit.

kraigher, 2017-08-07:
I updated the Vivado example just now according to the current best thinking.

bundi78, 2017-08-08:
thanks, i will look into it

closed 2017-12-22 by kraigher