VUnit/vunit #270: VUNIT IP

mohsenblueice, 2017-08-07:
Hi all,
I am trying to use VUNIT aling with XILINX IP cores. But I could not run the IP example. it returns various error which after spending lots of time I have not been able to fix it. does anybody have experience?

kraigher, 2017-08-07:
What errors? Could you post the output. I know we have not tested the example with the lasted Vivado so some parts may be outdated.

kraigher, 2017-08-07:
@mohsenblueice I updated the Vivado example according to the current best thinking. It was quite old. You can try again if you want.

mohsenblueice, 2017-08-08:
@kraigher  Thank you for your update.  I tried to run new scripts but the first error is:
"
C:\Python27>python C:\Users\taherim\Documents\TESTIP\generate_vivado_project.py
Traceback (most recent call last):
  File "C:\Users\taherim\Documents\TESTIP\generate_vivado_project.py", line 7, in <module>
    from vivado_util import run_vivado
  File "C:\Users\taherim\Documents\TESTIP\vivado_util.py", line 11, in <module>
    from vunit.simulator_factory import SIMULATOR_FACTORY
ImportError: cannot import name SIMULATOR_FACTORY "

kraigher, 2017-08-08:
Then you are using an older version of VUnit than the example was written for. You should update to the same commit as the latest example.

mohsenblueice, 2017-08-08:
I've update to vunit-hdl-V2.1.1 and ran it again but I have new error:

"
C:\Python27>python C:\Users\taherim\Documents\TESTIP\generate_vivado_project.py
vivado -mode batch -notrace -nojournal -nolog -source C:\Users\taherim\Documents\TESTIP\tcl\generate_project.tcl -tclargs C:\Users\taherim\Documents\TESTIP myproject
'vivado' is not recognized as an internal or external command,
operable program or batch file.
Traceback (most recent call last):
  File "C:\Users\taherim\Documents\TESTIP\generate_vivado_project.py", line 21, in <module>
    main()
  File "C:\Users\taherim\Documents\TESTIP\generate_vivado_project.py", line 18, in main
    run_vivado(join(root, "tcl", "generate_project.tcl"), root, "myproject")
  File "C:\Users\taherim\Documents\TESTIP\vivado_util.py", line 151, in run_vivado
    check_call(cmd, shell=True)
  File "C:\Python27\lib\subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'vivado -mode batch -notrace -nojournal -nolog -source C:\Users\taherim\Documents\TESTIP\tcl\generate_project.tcl -tclargs C:\Users\taherim\Documents\TESTIP myproject' returned non-zero exit status 1"

kraigher, 2017-08-08:
Then vivado is not in your PATH environment variable which is listed as a requirement in the readme.

mohsenblueice, 2017-08-08:
@kraigher You are right. But the point is that I am using vivado 2017 and there is no exe file to run vivado. there is a bath file to run vivado. could you please tell me what "PATH environment variable " should be?
"variable name = vivado" , "variable value =?"

kraigher, 2017-08-08:
Using `cmd.exe` on my machine I can set it like this:
```
set PATH=C:\Xilinx\Vivado\2016.4\bin;%PATH%
```

kraigher, 2017-08-08:
The PATH environment variable contains a list of directories to search for executables and executable scripts when typing a command name in the shell on both Linux and Windows. It is a standard mechanism.

kraigher, 2017-08-08:
https://en.wikipedia.org/wiki/PATH_(variable)

mohsenblueice, 2017-08-08:
OK. Now it generates myproject. but when I excute run.py there is an error which does not recognize modelsim while I am using Vuint  with modelsim (questa) without any difficulty. 

"Executing cmd 'C:\questasim64_10.6a\win64/vcom.exe -version -32' with output '.cxl.modelsim.version' (Tue Aug  8 10:51:46 2017)...
compile_simlib[exe]: The command line 'C:\questasim64_10.6a\win64/vcom.exe -version -32' returned exit code '0'.
ERROR: [Vivado 12-3754] Failed to find the 'modelsim' simulator executable. Make sure to set the 'modelsim' installation environment and retry this command to compile the libraries for this simulator. For more information on tool setup refer 'modelsim' user guide.
Library compilation for 'modelsim' ignored.
ERROR: [Common 17-39] 'compile_simlib' failed due to earlier errors.

    while executing
"compile_simlib -force \
    -language all \
    -simulator ${simulator_name} \
    -32bit  \
    -verbose  \
    -library unisim \
    -family  all \
..."
"

kraigher, 2017-08-08:
${simulator_name} should be `questa` in your case I think. The example will insert `modelsim` since VUnit uses modelsim interface for both modelsim and questa and does not differentiate the two. You can manually edit the TCL file to make it say questa.

kraigher, 2017-08-08:
If you have a 64-bit license and want to use it you should also manually edit that part of the `compile_standard_libs.tcl` file that says `-32bit` now.

mohsenblueice, 2017-08-08:
@kraigher I appreciate your complete and clear hints. 
It works now. finally I managed to run your example.

kraigher, 2017-08-08:
Nice to hear.

mohsenblueice, 2017-08-08:
@kraigher I tried to run my example but I understood that vivado has different behavior for different IPs.
for RAM and FIFO it generates a folder containing many files but for example for DDS it only generates a DDS_0.xcix file. Then when I execute the run.py ;

"C:\Python27>python C:\Users\taherim\Documents\Vivado\DCU_UT\DCU_UT.srcs\sources_1\imports\Vivado\RX\run.py  *TB_DDS_Wrapper*
Standard libraries already exists in C:\Users\taherim\Documents\Vivado\DCU_UT\DCU_UT.srcs\sources_1\imports\Vivado\RX\vivado_libs\standard, skipping
Traceback (most recent call last):
  File "C:\Users\taherim\Documents\Vivado\DCU_UT\DCU_UT.srcs\sources_1\imports\Vivado\RX\run.py", line 23, in <module>
    project_file=join(root, "..\\..\\..\\..\\..\\..\\DCU_UT", "DCU_UT.xpr"))
  File "C:\Users\taherim\Documents\Vivado\DCU_UT\DCU_UT.srcs\sources_1\imports\Vivado\RX\vivado_util.py", line 27, in add_vivado_ip
    compile_project_ip(vunit_obj, project_file, project_ip_path)
  File "C:\Users\taherim\Documents\Vivado\DCU_UT\DCU_UT.srcs\sources_1\imports\Vivado\RX\vivado_util.py", line 63, in compile_project_ip
    include_dirs=include_dirs if is_verilog else None)
  File "C:\Python27\lib\site-packages\vunit\ui.py", line 1148, in add_source_file
    self._library_name, abspath(file_name), preprocessors)
  File "C:\Python27\lib\site-packages\vunit\ui.py", line 639, in _preprocess
    code = ostools.read_file(file_name)
  File "C:\Python27\lib\site-packages\vunit\ostools.py", line 279, in read_file
    with io.open(file_name, "r", encoding=encoding) as file_to_read:
IOError: [Errno 2] No such file or directory: 'c:\\Users\\taherim\\Documents\\Vivado\\DCU_UT\\DCU_UT.srcs\\sources_1\\ip\\DDS_0\\hdl\\dds_compiler_v6_0_vh_rfs.vhd'"

I tried to copy missing files manually from Xilinx installation directory to expected directory but it needs many files not only one! any idea?

kraigher, 2017-08-08:
If you make any changes to the project such as add or remove ip you should remove vivado_libs/project_ip

mohsenblueice, 2017-08-08:
Anyway it is not vunit problem. So I close this issue. thanks again.

closed 2017-08-08 by mohsenblueice