VUnit/vunit #271: VIVADO IP Test

mohsenblueice, 2017-08-09:
Hi All,
I am using Vivado IP and vunit and there is an error which says that unknown extension file. any idea?
It is a FIR filter and it has many files which seems are not known for Vunit.
"
Traceback (most recent call last):
  File "C:\Users\taherim\Documents\DCU_Vivado\new\run.py", line 25, in <module>
    project_file=join(root, "myproject", "myproject.xpr"))
  File "C:\Users\taherim\Documents\DCU_Vivado\new\vivado_util.py", line 27, in add_vivado_ip
    compile_project_ip(vunit_obj, project_file, project_ip_path)
  File "C:\Users\taherim\Documents\DCU_Vivado\new\vivado_util.py", line 63, in compile_project_ip
    include_dirs=include_dirs if is_verilog else None)
  File "C:\Python27\lib\site-packages\vunit\ui.py", line 1141, in add_source_file
    file_type = file_type_of(file_name)
  File "C:\Python27\lib\site-packages\vunit\project.py", line 901, in file_type_of
    raise RuntimeError("Unknown file ending '%s' of %s" % (ext, file_name))
RuntimeError: Unknown file ending '.mif' of c:/Users/taherim/Documents/DCU_Vivado/new/myproject_ip/RxChannelFIR/RxChannelFIR.mif
"

kraigher, 2017-08-09:
A mif file is not an RTL file and should not be added to VUnit. If I remember correctly it is a memory initialization file that is programmatically read by the behavioral model during elaboration. It think it needs to be located in the working directory of the simulator during elaboration as the search path is hard coded that way. Not very automation friendly. Have a look at this old issue that was created by another user that needed to handle .mif files https://github.com/VUnit/vunit/issues/236

mohsenblueice, 2017-08-09:
@kraigher  the issue which you mentioned is different. for my case, vunit returns an error which "mif" is not a known extension file. it finds the file but it can not accept this file format.
the simulation has not started yet it is in the phase of adding IP:
 add_vivado_ip -> compile_project_ip ->  add_source_file -> ERROR

kraigher, 2017-08-09:
The first problem is that the mif file should not be added to via `add_source_file` at VUnit at all. You have to modify the example to ignore that file and not perform `add_source_file` on it or modify the Vivado tcl not not list mif files in the compile order output. 

Once you get past that point and actually start running a simulation you are going to run into the problem of the mif file being in your simulator working directory which is where the second issue is relevant. This is why I posted the issue, I was thinking one step ahead.

mohsenblueice, 2017-08-09:
"add_source_file"  is done in ui.py. do you mean I have to modify this file?
for another option "modify the Vivado tcl not not list mif files in the compile order output" can you tell me how?

kraigher, 2017-08-09:
`add_source_file` is not done (I assume you mean called) in `ui.py` it is called from the Vivado example and it is there you need to make the modification. You need to modify the Vivado example such that it does not call `add_source_file` with .mif files. You to that by filtering the list of files using Python programming to remove the elements that are .mif files. Alternatively it could be possible to make Vivado avoid including .mif files in the compile order output but I do not know how to do that so you would have to investigate. My feeling is that just ignoring the files in Python by modifying the vivado_util.py is probably easier.

closed 2017-12-22 by kraigher