VUnit/vunit #273: VIVADO IP

mohsenblueice, 2017-08-24:
Hi,
I am trying to fix problem with IP one after another.  
my current problem is: I have my project with some custom IP and then I "run.py". It compiles standard library and also my custom IPs and generates required simulation files.  but it returns this error:
"
# ** Fatal: (vsim-3729) Value ?(1095521093) of generic "DIFF_TERM" is out of range FALSE (0) to TRUE (1).
#    Time: 0 ps  Iteration: 0  Instance: /tb_txinputphy/U_UT/U_SerDes1/inst/pins[0]/ibufds_inst File: C:/Xilinx/Vivado/2017.1/data/vhdl/src/unisims/primitive/IBUFDS.vhd Line: 35
# FATAL ERROR while loading design
# Error loading design
Error loading design
"
any idea?

kraigher, 2017-08-24:
It says you set a generic out of bounds but you already seen that yourself I guess.

mohsenblueice, 2017-08-24:
Yes I see. but the point is that when I run this test without using Vunit there is no error. It means there is nothing wrong with the code. I think something is wrong when Vunit compiles libraries. I mean IP libraries.

kraigher, 2017-08-24:
Are you sure the same Vivado version has been used to compile unisim as the version of the IP?

I would also compare the list of involved vhd and v files between the situation when it is working and when it is not.

Could also be something with target language VHDL vs Verilog

mohsenblueice, 2017-08-24:
1- There is only one Vivado version ; 2017.1
2- The project I am working is simple with only two VHDL files; TOP and TB_Top and one IP core (SERDES). Now I do not understand which files do you mean to compare?
3-targert language in both situation is VHDL.

kraigher, 2017-08-24:
2. I mean which VHDL and Verilog files make up the simulation model of the IP. Are the same IP files compiled in both the working and non working situation. If the exact same files using the exact same unisim library is in effect in both situations there can be no difference.

mohsenblueice, 2017-08-24:
Ok let me explain more ;
- I added my IP manually to the folder of myproject_ip
- then I generated a project using "generate_vivado_project.py" 
-then I used "run.py"
It generated everything according to my observation! but it failed because of the mentioned error.
THEN:
-I removed Vunit stuff form test bench file. (library declaration and generic definition)
- I ran Vivado manually and opened the myproject.xpr which was produced using "generate_vivado_project.py" .
- I ran simulation in Vivado. it worked.

kraigher, 2017-08-24:
The way forward is to compare the compile operations performed by Vivado (files, flags, order) with what happens when using. You could get Vivado to generate a compile.do file for modelsim and run/inspect that for instance.

closed 2017-12-22 by kraigher