VUnit/vunit #274: VUnit python interface: "getting" a Library

Aaron-Hartwig, 2017-09-06:
# Problem
The [vunit.ui.VUnit.library](https://vunit.github.io/python_interface.html#vunit.ui.VUnit.library) method returns [new instances of a library](https://github.com/VUnit/vunit/blob/b5b9b2498a9150632b3ea0f8f701526fdc5f1c32/vunit/ui.py#L416). I understand that this is just an interface into the underlying `vunit.project.Library` class, but this becomes an issue when trying to access previously added libraries on an instance of VUnit programatically. For a bit more context, I am setting up a larger test suite using the pattern @LarsAsplund lays out on [SO](https://stackoverflow.com/questions/42980036/how-to-combine-multiple-vunit-run-py-files-into-a-single-vunit-run). When doing this I need to "get" Libraries I may have previously created and added files to so I can add additional files.

# Reproduction
```python
from vunit import VUnit
from os.path import dirname, abspath, join

# create the file we try to add
open('file.vhd', 'a').close()

proj_root = dirname(abspath(__file__))
vu = VUnit.from_argv()

try:
    test_lib = vu.library('Test')
except KeyError:
    print('Test Library does not exist, creating it now.')
    test_lib = vu.add_library('Test')

test_lib_2 = vu.library('Test')

print('Test Library returned after initial creation       : {0} - {1}'.format(test_lib.name, test_lib))
print('Test Library returned after "getting" it from VUnit: {0} - {1}'.format(test_lib_2.name, test_lib_2))

test_lib.add_source_file(join(proj_root, 'file.vhd'))
test_lib_2.add_source_file(join(proj_root, 'file.vhd'))

# Cannot access file from because we it now exists in two "libraries"
test_lib.get_source_file('file.vhd')
```

Aaron-Hartwig, 2017-09-06:
Looks like the file gets added to the underlying source files again, even if it does not get added to the library again. Is this desired behavior? https://github.com/VUnit/vunit/blob/b5b9b2498a9150632b3ea0f8f701526fdc5f1c32/vunit/project.py#L142

kraigher, 2017-09-06:
Sure it results in a invalid state. At least a warning is emitted. Maybe the warning should be raising an exception instead.

kraigher, 2017-09-06:
Now an exception is raised instead when adding a duplicate file.

closed 2017-09-06 by kraigher

Aaron-Hartwig, 2017-09-06:
@kraigher Looking at it from the user's perspective, I would rather log that VUnit saw a duplicate file attempted to be added and then not add it rather than raise an exception.  If an exception is raised, then the user still has to handle the exception anyway (which was my workaround to this issue in the first place). I discovered the invalid state because I was dynamically added libraries (and files to those libraries) to test suites without knowing what was added previously. To me, one of the advantages of VUnit was that it handled the case of me trying to add files that already existed in a library elegantly (log a warning, ignore file).

kraigher, 2017-09-06:
I prefer the exception for two reasons:
1. Would it not be a goal of the user to run the `run.py` file without warnings anyway?
2. If allowing the program to continue after adding a duplicate file we must define the behavior.
In the case of verilog if a previously added file already has other defines or include_paths set it could be a conflict.

commit f1ca1dbe (2017-09-06): Error on adding duplicate source files, closes #274