VUnit/vunit #277: ghdl external library

slaweksiluk, 2017-09-20:
Hello, I'm trying to use vunit together with my external precompiled library astri_hdl_lib and I encountered the problem. File common_pkg.vhd is part of astri_hdl_lib and is included inside axis_test_env_pkg.vhd which is included in tb_example.vhd. Despite astri_hdl_lib is precompiled, the errors are printed, however the specific ghdl command used is not printed. I suppose the problem is with passing additional compile option to ghdl, as both of this errors are disabled by 
"--ieee=synopsys", "-frelaxed-rules" switches.

The others strange behaviour is reporting the simulation status despite compilation failure. I use the latest vunit from repo master branch.
```

sim/>  python run.py -v -k --clean
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/vhdl/src/lang/lang.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/string_ops/src/string_ops.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/path/src/path.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/logging/src/log_types.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/logging/src/log_formatting.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/logging/src/log_special_types200x.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/logging/src/log_base_api.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/logging/src/log_base.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/logging/src/log_api.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/logging/src/log.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/core/src/stop_api.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/core/src/stop_body_2008.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/core/src/core_pkg.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/check/src/check_types.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/check/src/check_special_types200x.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/check/src/check_base_api.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/check/src/check_base.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/check/src/check_api.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/dictionary/src/dictionary.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/run/src/run_types.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/run/src/run_special_types200x.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/run/src/run_base_api.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/run/src/run_base.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/run/src/run_api.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/vunit_run_context.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/vunit_context.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/run/src/run.vhd into vunit_lib ...
Compiling ../../../../../../../usr/lib/python2.7/site-packages/vunit_hdl-2.1.2rc0-py2.7.egg/vunit/vhdl/check/src/check.vhd into vunit_lib ...
Compiling tb_example.vhd into lib ...
Running test: lib.tb_example.all
Running 1 tests

Starting lib.tb_example.all
axis_test_env_pkg.vhd:126:10: unit "std_logic_misc" not found in library "ieee"
common_pkg.vhd:20:10: unit "std_logic_misc" not found in library "ieee"
common_pkg.vhd:25:17: type of a shared variable must be a protected type
common_pkg.vhd:26:17: type of a shared variable must be a protected type
/usr/bin/ghdl: compilation error
fail (P=0 S=0 F=1 T=1) lib.tb_example.all (0.0 seconds)

==== Summary ==============================
fail lib.tb_example.all (0.0 seconds)
===========================================
pass 0 of 1
fail 1 of 1
===========================================
Total time was 0.0 seconds
Elapsed time was 0.0 seconds
===========================================
Some failed!
sim/>  

```
TB:
```
library vunit_lib;
context vunit_lib.vunit_context;
library astri_hdl_lib;
use astri_hdl_lib.axis_test_env_pkg.all;

entity tb_example is
  generic (runner_cfg : string);
end entity;

architecture tb of tb_example is
begin
  main : process
  begin
    test_runner_setup(runner, runner_cfg);
    report "Hello world!";
    test_runner_cleanup(runner); -- Simulation ends here
  end process;
end architecture;
```

```
from vunit import VUnit

# Create VUnit instance by parsing command line arguments
vu = VUnit.from_argv()

#vu.add_external_library("astri_hdl_lib", "../../astri_hdl_lib")

# Create library 'lib'
lib = vu.add_library("lib")

# Add all files ending in .vhd in current working directory to library
lib.add_source_file("tb_example.vhd")

vu.set_compile_option("ghdl.flags", ["--ieee=synopsys", "-frelaxed-rules"])

# Run vunit function
vu.main()
```

Example command to compile my library object
`ghdl -a --work=astri_hdl_lib --ieee=synopsys --workdir=/home/lab-buddy/Documents/AGGA-4/HDL-LIB/astri_hdl_lib/ --std=08 -frelaxed-rules wbs_reg_user_pkg.vhd	`

kraigher, 2017-09-20:
It is probably not a compilation error but an error that first occurs during elaboration/simulation. You might need to set the correct flags to GHDL elab_run command via VUnit sim_options. GHDL often requires flags both when compiling/analyzing and elaborating/simulating.

slaweksiluk, 2017-09-21:
Exactly, I forgot about two stage ghdl compilation flow. Thank you for help. By the way, is there an option to print vunti ghdl elaboration commands? The '-v' switch doesn't provide such an information which would help me find the mistake faster.

LarsAsplund, 2017-09-21:
You can use the `--log-level` option.

closed 2017-12-08 by kraigher

dstadelm, 2019-03-22:
> Exactly, I forgot about two stage ghdl compilation flow. Thank you for help. By the way, is there an option to print vunti ghdl elaboration commands? The '-v' switch doesn't provide such an information which would help me find the mistake faster.

Have you ever found out what the problem was. I am running into the same problem that the std_logic_misc is not found in library ieee.

Regards
Dave