VUnit/vunit #278: Circular reference error thrown due to module name occuring within a commented part of a .v file

adrspa, 2017-09-25:
I'm compiling a large set of code with both VHDL and Verilog files when I run into the "circular reference" error. Basically I have something like this:

In file1.v:
module TVSENSE
…

In file2.v
module TVSENSE_NET_FM  

…

if ($test$plusargs ("TVSENSE_SHORT_CONVERSION")) begin
    shortConversion = 1'b1;
    $display ("\n*************************************************\
TVSENSE NOTE %m at time %f ns\
   You are using TVSENSE model in a FAST (SHORT) CONVERSION MODE\
\
   Please note: THIS IS SIMULATION ONLY feature\
                it does not exist in real hardware !!!!!!!!!!\
\
   Users MUST run simulation with FULL CONVERSION CYCLES as well !!!!!!!!!!\
\
*************************************************\
", $realtime, conversion_length, conversion_length, conversion_length + 9);

It seems that the parser doesn't see that TVSENSE is within a commented part of the Verilog code. It works if I change this line:
TVSENSE NOTE %m at time %f ns\
to this:
TVSENSE2 NOTE %m at time %f ns\

Is this a bug in how VUnit parses .v files?

kraigher, 2017-09-26:
From your example it seems that TVSENSE is not within a commented part of the code but rather a string literal. It seems the code contains a multi line string literal. I do not know if that is legal Verilog without escaping the new-lines inside the string quotes with backslash. Modelsim gives a warning with multi line string literas. Anyway the VUnit tokenizer does not handle multi line strings causing an incorrect tokenization. The dependency scanning works on the token stream and ignores string literals and comments but a multi line string causes the token stream to be incorrect in this case.

kraigher, 2017-09-26:
You can try the commit I referenced in this issue on the latest master branch. It should tokenize multi line strings now.

kraigher, 2017-09-29:
It is now part of 2.2.0 release.