VUnit/vunit #280: Modelsim: Invalid value for generic "output_path".  Not using this value.

dave2pi, 2017-10-26:
Since commit [76d17d5](https://github.com/VUnit/vunit/commit/76d17d5f85d24de7686c67431e126972228886fc) "Added human readable test output paths. #211" `config_name`'s containing commas can be passed to Modelsim as part of the top level generic `output_path` and Modelsim doesn't seem to like commas.

If I run the vunit example from `vunit\examples\vhdl\generate_tests` (and I run it from C:\ to avoid any windows path length issues) I'm getting the following error:
I'm running ModelSim PE 10.5c Jul 21 2016
(output abbreviated)
```bash
C:\generate_tests>python run.py --version
2.2.0
C:\generate_tests>python run.py --clean
Compiling...
Starting lib.tb_generated.data_width=1,sign=False.Test 1
# vsim -modelsimini C:/generate_tests/vunit_out/modelsim/modelsim.ini -wlf C:/generate_tests/vunit_out/test_output/lib.tb_generated.data_width=1,sign=False.Test_1_b8881b8ddd543fcd90d8901c4daae75a83945267/modelsim/vsim.wlf -quiet -t ps -onfinish stop lib.tb_generated(a) -L vunit_lib -L lib -g/tb_generated/runner_cfg="active python runner : true,enabled_test_cases : Test 1,output path : C::/generate_tests/vunit_out/test_output/lib.tb_generated.data_width=1,,sign=False.Test_1_b8881b8ddd543fcd90d8901c4daae75a83945267/,tb path : C::/generate_tests/test/" -g/tb_generated/output_path=C:/generate_tests/vunit_out/test_output/lib.tb_generated.data_width=1,sign=False.Test_1_b8881b8ddd543fcd90d8901c4daae75a83945267/ -g/tb_generated/sign=False -g/tb_generated/data_width=1 -g/tb_generated/message=set-for-entity
# Start time: 09:24:39 on Oct 26,2017
# //  ModelSim PE 10.5c Jul 21 2016
# ** Warning: (vsim-3351) Invalid value "C:/generate_tests/vunit_out/test_output/lib.tb_generated.data_width=1,sign=False.Test_1_b8881b8ddd543fcd90d8901c4daae75a83945267/" for generic "output_path".  Not using this value.
#    Time: 0 ps  Iteration: 0  Instance: /tb_generated File: C:/generate_tests/test/tb_generated.vhd
# ** Fatal: (vsim-3350) Generic "output_path" has not been given a value.
#    Time: 0 ps  Iteration: 0  Instance: /tb_generated File: C:/generate_tests/test/tb_generated.vhd Line: 15
# FATAL ERROR while loading design
# Error loading design
Error loading design
# End time: 09:24:41 on Oct 26,2017, Elapsed time: 0:00:02
# Errors: 1, Warnings: 1
fail (P=0 S=0 F=1 T=9) lib.tb_generated.data_width=1,sign=False.Test 1 (4.9 seconds)
```

If I change the config_name formatting in `generate_tests/run.py` from:
`config_name = "data_width=%i,sign=%s" % (data_width, sign)`
to
`config_name = "data_width=%i.sign=%s" % (data_width, sign)`
without a comma, then modelsim accepts it.

Alternatively if I edit `vunit/test_runner.py` to include comma as an illegal char:
`ILLEGAL_CHARS = ' <>"|:*%?\\/#&;(),'`
it also now works.

I noticed this issue when I tried to rerun some test sets I wrote back in June in which I was formatting the config_name in the same comma seperated style as the generate_tests example. Replacing `,` with `.` in these fixes them as well.

VUnit 2.0.2rc0 works fine since the paths only contain a hash
VUnit 2.2.0 doesn't work
VUnit 2.2.1rc0 doesn't work

Has anybody else noticed this?

kraigher, 2017-10-26:
You can add comma to the list of ILLEGAL_CHARS as you suggested such that it will be replaced in the path name. Can you make a pull request out of it?

dave2pi, 2017-10-30:
I thought it was neater to fix in modelsim_interface.py. Seems there are weird constraints on the braces/brackets. Through some experimentation I've worked out:

Strings with a space require braces and quotes eg. `'{"x y"}'` or `'{"x, y"}'`.
String with a comma but no space must have quotes but no braces eg. `'"x,y"'`.
String with neither comma or space require no braces or quotes eg. `'xy'`.

Something like `-g/tb/output_path={"x,y"}` in common.do won't expand out properly for some reason. 

I've added pull request #284.

commit a18bca05 (2017-10-30): Fixes commas in Modelsim generics. Closes #280. (#284)

Modelsim interface needs to enclose strings for generic values containing spaces and commas in quotes or quotes and braces. 
Strings with a space require braces and quotes eg. `'{"x y"}'` or `'{"x, y"}'`. 
String with a comma but no space must have quotes but no braces eg. `'"x,y"'`.
String with neither comma or space require no braces or quotes eg. `'xy'`.
closed 2017-10-30 by kraigher