VUnit/vunit #281: Does add_config support generic type time?

ttobsen, 2017-10-26:
Hello VUnit Devs,

I'm trying to automate a testbench where I have several generics of type time. Is it possible to handle this generic type using add_config and dict in some kind of VHDL style? E.g.

`testbench.add_config(name="example", generics=dict(SIZE=1000, EX_TIME=???))`

I tried using a string for the time content, e.g. `EX_TIME="0.1 us"` but this is not working. When using simple integers, e.g. `EX_TIME=5` the content is interpreted with the simulation resolution (which is some kind of expected I think).

So the question is, if there is a more VHDL kind of time representation which can be used for the different configurations? This would absolutely improve the readability of the python run script.

Thanks for the support and keep on with the great VUnit work!

Best Regards,
Tobi

kraigher, 2017-10-26:
Many simulators do not support setting a time generic. The best method is to use a VHDL string as the generic and use time'value in the test bench to convert to a time data type.

ttobsen, 2017-10-26:
Wow, this was fast. Thanks for your support, it's working like a charm.

Maybe this should find its way into the documentation (I hope I didn't miss this).

kraigher, 2017-10-26:
Yes documenting this could be valuable. Pull requests improving the documentation is welcome.

ttobsen, 2017-10-26:
That's perfect. Because I'm pretty weak in Python, I can at least give something by improving documentation. You'll get a PR when I'm done. Thanks!

closed 2017-12-22 by kraigher