VUnit/vunit #282: Recursive file path

eschmidscs, 2017-10-27:
Hi

Is there a way to add files recursively from a starting directory?
I'm not experienced with Python, so maybe that works by replacing join with something?

Thanks and best regards,
emanuel

kraigher, 2017-10-27:
This is really a generic Python question. Remember a run.py file is really just a Python script. The Python language is one of the most used programming languages in the world with a huge amount of information online. Googling "python add files recursively" will give you plenty of different ways to achieve what you want. Hope this was helpful.

"give a man a fish and you feed him for a day; teach a man to fish and you feed him for a lifetime"

eschmidscs, 2017-10-30:
Ok, thanks for teaching how to to fish ;)
It didn't help that I didn't read "... or a list of files" in the add_source_files documentation.

closed 2017-10-30 by eschmidscs