VUnit/vunit #283: Proposals for a better overview in the terminal

asicnet, 2017-10-27:
Hi
I have two proposals for a better overview in the terminal. Maybe it is interesting.

1. In the compile phase the output is not divided in columns. 
I changed this to 
Compiling  into <library>           <path to file>
Compiling  into vunit_lib            ..\..\..\..\..\VUnit\vunit\vunit\vhdl\path\src\path.vhd
Compiling  into osvvm               ..\..\..\..\..\VUnit\vunit\vunit\vhdl\osvvm\VendorCovApiPkg.vhd

2. Added the option --log-only
If a lot of test are failed I have to much output in the terminal or command window. That is unclear.
The option suppresses the output and print instead the path to the output.txt.

If this is interesting, i have the files attached.
Changes of 4 files of version 2.2.0
[vunit.zip](https://github.com/VUnit/vunit/files/1423141/vunit.zip)

best regards
AsicNet

Example:
-----------------------------------------------------------------------------------------
Compiling  into tb_lib               tb\test_tb.vhd

Starting tb_lib.test_tb.t1_t2
Logfile: E:\Project\test\vunit_out\test_output\\tb_lib.259304bfcbf4910ed9\output.txt
fail (P=2 S=0 F=2 T=3) tb_lib.test_tb.t1_t2 (3.9 seconds)

Starting tb_lib.test_tb.t1_t3
Logfile: E:\Project\test\vunit_out\test_output\\tb_lib.7845hfiehf78juh4\output.txt
fail (P=3 S=0 F=2 T=3) tb_lib.test_tb.t1_t3 (3.7 seconds)
==== Summary ===============
fail tb_lib.test_tb.t1_t1 (3.8)
fail tb_lib.test_tb.t1_t2 (3.9) 
fail tb_lib.test_tb.t1_t3 (3.7) 
===========================
pass 0 of 3
fail 3 of 3
====================================================================================================================
Total time was 11.9 seconds
Elapsed time was 12.3 seconds
====================================================================================================================
Some failed! 

[vunit.zip](https://github.com/VUnit/vunit/files/1423141/vunit.zip)

kraigher, 2017-11-01:
If you publish this in a fork and make a pull request it is easier for us to try it.

kraigher, 2017-11-01:
Maybe --log-only should be called -q/--quiet

asicnet, 2017-11-02:
Hi
I added the files and create a branch, but i dont know if it was correct.

kraigher, 2017-11-07:
I posted a few comments on your commit. The most convenient way for us is if you make a pull request out of it then we can just press merge if it meets quality standards and is a valid feature.

kraigher, 2017-11-07:
I fixed the compile printouts myself now. Also added a colored status printout after compiling all files.

kraigher, 2017-11-08:
I have also implemented the -q/--quiet flag.

closed 2017-11-08 by kraigher

asicnet, 2017-11-09:
Hi
today i test your changes.
I like it.  👍  -q is my standard and if errors occur i look quickly into the log-files

Maybe it becomes standard and the verbose option get a optional verbosity level  --verbose [vunit, fail, all,...]    :-)