VUnit/vunit #286: get_implementation_subset dependency fail on components

gsorrenti, 2017-11-07:
In some particular situation get_implementation_subset fail to detect correctly the dependency tree. It seems to be related to instantiation by components.
Please find bug test on [my forch](https://github.com/gsorrenti/vunit/tree/master/examples/vhdl/get_implementation_subset_issue).

The conditions are:
1. circular_dependency
The same entity is defined in two different libraries and inside each library it is instantiated by component in others entities. Vunit raise an error of circular dependency.
1. multiple_library_dep
The same entity is compiled in two different libraries. The dependency is propagated to all libraries even if only one should be involved.
1. separated_architecture
In case of entity and architecture definition located in different files, the dependency is not propagated to architecture files.
1. verilog_component
In case of verilog entities instantiated by components, the dependency is not propagated.

kraigher, 2017-11-07:
It would be helpful if you could formulate your reproducing cases as test cases in this file: https://github.com/VUnit/vunit/blob/master/vunit/test/unit/test_project.py

If you want to also try to fix the implementation that is also welcome.

gsorrenti, 2017-11-09:
Ok @kraigher, I'll add test cases as soon as possible.

gsorrenti, 2017-11-13:
Sorry @kraigher I cannot figure how to use a project defined in test_project.py. Sure it's my python competence lack.

gsorrenti, 2017-11-13:
Ok, one step ahead. I understood I have to run tox.

gsorrenti, 2017-11-13:
I pushed on my fork the four new tests. Do you prefer a pull request?

kraigher, 2017-11-13:
A pull request is convenient. 

BTW: You do not need to use tox to run tests. Just do:
```
python -m pytest vunit/test/unit/
```

Maybe need to install pytest first:
```
python -m pip install pytest mock
```

kraigher, 2017-12-22:
@gsorrenti I made fixes such that your tests pass now. Also fixed such that architecture files are added for component instantiations as well as entity instantiations. Thank you for contributing the test cases.

closed 2017-12-22 by kraigher

gsorrenti, 2017-12-27:
You are welcome

gsorrenti, 2018-01-30:
I experimented some other problem related to component packages. As soon as possible I will provide new tests.

commit 1392b938 (2017-12-22): Merge branch 'issue_286'. Closes #286