VUnit/vunit #290: Idea for easier use of composite generics

sthenc, 2018-01-04:
Hello,

I just reworked an old testbench to work with composite generics, and I use something like this:

```
    -- ternary operator
    function sel(cond: boolean; if_true, if_false: string) return string is
    begin
        if (cond = true) then
            return(if_true);
        else
            return(if_false);
        end if;
    end function sel; 

    function get_or_default(dict: string; key: string; default_value: string) return string is
    begin
        return sel(has_key(dict, key), get(dict, key), default_value);
    end function sel; 
    
    constant STRIPE      : integer   := integer'value(get_or_default(video_parameters, "STRIPE", "8192"));
```

Would you consider adding a similar function to the dictionary module?
[dictonary.vhd](https://github.com/VUnit/vunit/blob/master/vunit/vhdl/dictionary/src/dictionary.vhd)

kraigher, 2018-01-04:
Good idea, there is a similar feature in Python. If you want to make a pull request it yourself look into these files:

Add new test here:
https://github.com/VUnit/vunit/blob/master/vunit/vhdl/dictionary/test/tb_dictionary.vhd

Add new feature here:
https://github.com/VUnit/vunit/blob/master/vunit/vhdl/dictionary/src/dictionary.vhd

sthenc, 2018-01-04:
Ok, I'll give it a shot.

closed 2018-01-04 by kraigher