VUnit/vunit #292: bfm ghdl compile error

slaweksiluk, 2018-01-04:
Hello, I would like to try bfm built in vunit as it looks very promising. I tried to run /vunit/vunit/vhdl/verification_components/run.py at bfm branch, but I got following error:
```
Compiling into vunit_lib: ../../../../../../../../usr/local/lib/python2.7/dist-packages/vunit_hdl-2.3.1rc0-py2.7.egg/vunit/vhdl/logging/src/log_types.vhd                              failed
=== Command used: ===
/home/slawek/install_prefix/bin/ghdl -a --workdir=/home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/ghdl/libraries/vunit_lib --work=vunit_lib --std=08 -P/home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/ghdl/libraries/vunit_lib -P/home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/ghdl/libraries/osvvm /usr/local/lib/python2.7/dist-packages/vunit_hdl-2.3.1rc0-py2.7.egg/vunit/vhdl/logging/src/log_types.vhd

=== Command output: ===
/usr/local/lib/python2.7/dist-packages/vunit_hdl-2.3.1rc0-py2.7.egg/vunit/vhdl/logging/src/log_types.vhd:12:10: unit "lang" not found in library "vunit_lib"
/home/slawek/install_prefix/bin/ghdl: compilation error
```
```
verification_components> ghdl -v
GHDL 0.36-dev (v0.35-45-ge9f5de3) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
I've juest checked out bfm branch. Install command was:
```
sudo python setup.py install
```

Full log
[errlog.txt](https://github.com/VUnit/vunit/files/1604770/errlog.txt)
Please let me know if additional information is needed

kraigher, 2018-01-04:
I think there was a problem with installing from the branch. The file that gives the error is deleted on the branch and should not be in the installed folder. The setup.py installer should remove old files, maybe it doesnt, in that case see: https://gist.github.com/myusuf3/933625

kraigher, 2018-01-04:
When I experiment with different work in progress when developing VUnit I never install it for every change I make. Instead I set the PYTHONPATH environment variable to the root of the VUnit repository and it will import from there directly.

slaweksiluk, 2018-01-04:
Yes, working on development version is definitely more convenient with PYTHONPATH set. Command which runs all the tests:
```
PYTHONPATH=/home/slawek/git/github/vunit python run.py
```
Thank you for quick answer!

kraigher, 2018-01-04:
Great. If you have any feedback on the verification components please let us know via for example Gitter. 

We will add the verification components to the master branch in a while as a beta version. This means we want people to try it but still keep our hands free to make some breaking changes as we get more input.

closed 2018-01-04 by kraigher