VUnit/vunit #294: SystemVerilog support

alinaivanovaoff, 2018-01-10:
Hi!

Could you clarify, please, what part of SystemVerilog is supported by VUnit or where I can check it?
My company uses SystemVerilog and we worry what we cannot use VUnit because of it.

Best regards,
Alina ivanova

LarsAsplund, 2018-01-10:
As long as the simulator is supported by VUnit and the testbench  is on the format described [here]( user_guide.html#systemverilog-test-benches) you can use VUnit. What parts of SystemVerilog you can use to describe the rest of your testbench depends on what your simulator supports.

alinaivanovaoff, 2018-01-10:
Thanks for quick answer. I just confused about "[SystemVerilog (Support is experimental)](https://vunit.github.io/about.html#languages)"

LarsAsplund, 2018-01-10:
The reason for "experimental" is that the main developers do not use SystemVerilog professionally so we have limit personal experiences. We, like most  of the VUnit community, use VHDL but there are also SystemVerilog users out there and we haven't had many complaints. The Python part of VUnit which is responsible for handling the simulator is mostly HDL language independent. It doesn't care about what language is driving the simulation. So most of the things used to drive a SystemVerilog testbench is also used when using VUnit with VHDL.

If you're willing to give it a try we would be happy to take any feedback you may have. You can use this forum or our [chat](https://gitter.im/VUnit/vunit).

LarsAsplund, 2018-01-12:
@alinaivanovaoff I'm closing this issue now but you may also be interested in the work to add VUnit support for the free Icarus Verilog simulator. 
 For more information see #188.

closed 2018-01-12 by LarsAsplund