VUnit/vunit #296: add_source_files *.svh

alinaivanovaoff, 2018-01-12:
Hi guys!

I try to include into library *.svh file - it's SystemVerilog header.
 lib.add_source_files("../../common/rtl/*.svh") 
I got an error:
RuntimeError: Unknown file ending '.svh' of ../../common/rtl\com_rtl_mcr.svh

kraigher, 2018-01-12:
Header files are not added to the project. They are included by other files. Just like with manual compilation a header file is never passed to the simulator directly but included by the preprocessor when compiling the file that includes it. 

If you really want to force VUnit to add it as a regular SystemVerilog file that is supposed to be compiled instead of included you can use the file_type argument to add_source_files. I suspect this is no twhat you want.

NOTE: You need to set the include_dir when adding your other files that includes this header.

1:  http://vunit.github.io/python_interface.html#vunit.ui.VUnit.add_source_files

alinaivanovaoff, 2018-01-12:
Thanks for quick reply, but I a bit misunderstood. If I write run.do for Modelsim from scratch, I include such line: 
set INCLUDE_PATH +incdir+../../common/rtl/+../../common/tb/+../rtl/ - it's location for headers 

when I include source file, I use this path like:
vlog $INCLUDE_PATH -sv -lint ../../common/tb/com_tst_pkg.sv

But what should I do in the run.py? 
I got such error:
** Error: D:\Projects\rebuf_cat_mem\vunit_test\tb\vunit_test_tb.sv(20): Cannot open `include file "C:\Program Files\Python36\lib\site-packages\vunit\verilog\include/com_tst_mcr.svh".

run.py doesn't know where my *.svh files are located

kraigher, 2018-01-12:
You need to add all your include search paths to the `include_dirs` argument of `add_source_files`.
This corresponds to setting the `+incdir` argument in vlog.

```python
INCLUDE_DIRS = ["../../common/rtl/",
                "../../common/tb/", ...]
lib.add_source_files(..., include_dirs=INCLUDE_DIRS)
```

alinaivanovaoff, 2018-01-12:
Thank you. It works.

closed 2018-01-12 by alinaivanovaoff