VUnit/vunit #297: * Error: ** while parsing macro expansion: 'TEST_SUITE'

alinaivanovaoff, 2018-01-12:
** Error: ** while parsing macro expansion: 'TEST_SUITE' starting at D:\\vunit_tb.sv(16)

testbench code:

`timescale 1 ns / 1 ps
`include "vunit_defines.svh"
module vunit_test_tb  ();
    parameter CLOCK                          = 12.5;

    var clk;
    logic rstn;
    logic [15:0] data_in;
    initial begin
        clk = 1;
        forever begin
            #(CLOCK / 2.0) clk               = ~clk;
        end
    end

    `TEST_SUITE begin
        `TEST_SUITE_SETUP begin
            initial begin
                $display("Running test suite setup code");
                rstn = 0;
                data_in = 0;
                #10;
                rstn = 1;
            end
        end

        `TEST_CASE("Test_pass") begin
            initial begin
                $display("This test case is expected to pass");
                @(posedge rstn);
                `CHECK_EQUAL(data_in, 0);
            end
        end

        `TEST_CASE("Test_fail") begin
            initial begin
                $display("This test case is expected to fail");
                @(posedge rstn);
                `CHECK_EQUAL(data_in, 10);
            end
        end
    end;

    `WATCHDOG(1ns);
endmodule

Python:

from vunit import VUnit

# Create VUnit instance by parsing command line arguments
vu = VUnit.from_argv()

# Create library 'lib'
lib = vu.add_library("lib")

INCLUDE_DIRS = ["../../common/rtl/",
                "../../common/tb/"]

# Add all files ending in .vhd in current working directory to library
lib.add_source_files("../tb/vunit_tb.sv", include_dirs=INCLUDE_DIRS)

# Run vunit function
vu.main()

kraigher, 2018-01-12:
The TEST_SUITE macro itself is one process and expands to an `initial begin`. Thus you cannot define processes within the TEST_SUITE macro. If you need more than one process those are created outside the TEST_SUITE macro.

kraigher, 2018-01-12:
You can think of your TEST_SUITE as the master process of the test bench that controls all other processes.

alinaivanovaoff, 2018-01-12:
Please, try without initial. I have the same error without initial

kraigher, 2018-01-12:
https://github.com/VUnit/vunit/blob/master/vunit/verilog/include/vunit_defines.svh

kraigher, 2018-01-12:
Did you also remove the extra begin/end?

alinaivanovaoff, 2018-01-15:
Thank you @kraigher! I see.

alinaivanovaoff, 2018-01-15:
Hi @kraigher,
What is the problem with this code? 
I have such error: 
** Error: ** while parsing macro expansion: 'TEST_SUITE' starting at D:\Projects\rebuf_cat_mem\vunit_test\tb\vunit_tb.sv(19)

Code: 
`timescale 1 ns / 1 ps
//-----------------------------------------------------------------------------
`include "vunit_defines.svh"
//-----------------------------------------------------------------------------
module vunit_test_tb  ();
    parameter CLOCK                          = 12.5;
//-----------------------------------------------------------------------------
// Clock
    var clk;
    logic rstn;
    logic [15:0] data_in;
    initial begin
        clk = 1;
        forever begin
            #(CLOCK / 2.0) clk               = ~clk;
        end
    end

    `TEST_SUITE begin
        `TEST_SUITE_SETUP begin
            $display("Running test suite setup code");
            rstn = 0;
            data_in = 0;
            #10;
            rstn = 1;
        end

        `TEST_CASE("Test_pass") begin
            $display("This test case is expected to pass");
            `CHECK_EQUAL(data_in, 0);
        end

        `TEST_CASE("Test_fail") begin
            $display("This test case is expected to fail");
            `CHECK_EQUAL(data_in, 10);
        end
    end;

    `WATCHDOG(1ns);

    /*initial begin
        $display("Running test suite setup code");
        rstn = 0;
        data_in = 0;
        #10;
        rstn = 1;
        $display("This test case is expected to pass");
        $display("This test case is expected to fail");
    end*/
endmodule

Python:
from vunit import VUnit

# Create VUnit instance by parsing command line arguments
vu = VUnit.from_argv()

# Create library 'lib'
lib = vu.add_library("lib")

lib.add_source_files("../tb/vunit_tb.sv")

# Run vunit function
vu.main()

kraigher, 2018-01-15:
Maybe no semi-colon after end in some places? The algorithm for finding such problems is bisection; Comment out code until it works then uncomment until it fails in smaller and smaller steps. Quickly you will find only one small part of the code that causes the failure.

alinaivanovaoff, 2018-01-15:
Semi-colon after ends - is very strange SystemVerilog syntax. According this example https://github.com/VUnit/vunit/blob/master/examples/verilog/user_guide/tb_example.sv semicolon set just here `TEST_SUITE begin end;
I also has semicolon here. It's very weird, but https://github.com/VUnit/vunit/blob/master/examples/verilog/user_guide/tb_example.sv - gives the same error.

kraigher, 2018-01-15:
What version  of Modelsim do you have?

kraigher, 2018-01-15:
Can you run  the VUnit example without modification?

alinaivanovaoff, 2018-01-15:
ModelSim SE-64 10.4, Revision: 2014.12, Date: Dec 3 2014

alinaivanovaoff, 2018-01-15:
I already did it. 
``include "vunit_defines.svh"

module vunit_tb;
   `TEST_SUITE begin
      // Note: Do not place any code here (unless you are debugging
      // VUnit internals).

      `TEST_SUITE_SETUP begin
         // Here you will typically place things that are common to
         // all tests, such as asserting the reset signal and starting
         // the clock(s).
         $display("Running test suite setup code");
      end

      `TEST_CASE_SETUP begin
     // By default VUnit will run each test separately. However,
     // advanced users may want to run tests consecutively rather
     // than in separate instances of the HDL-simulator. In that
     // case the code placed in a TEST_CASE_SETUP block should
     // restore the unit under test to the state expected by the
     // test cases below. In many cases this block would only
     // assert/deassert the reset signal for a couple of
     // clock-cycles.
     //
     // When trying out VUnit for the first time this section
     // should probably be left empty.
         $display("Running test case setup code");
      end

      `TEST_CASE("Test that a successful test case passes") begin
         $display("This test case is expected to pass");
         `CHECK_EQUAL(1, 1);
      end

      `TEST_CASE("Test that a failing test case actually fails") begin
         $display("This test case is expected to fail");
         `CHECK_EQUAL(0, 1, "You may also optionally add a diagnostic message to CHECK_EQUAL");
     // Note: A test case will also be marked as failing if the
     // simulator stops for other reasons before the end of the
     // TEST_SUITE block is reached. This means that you don't
     // need to use CHECK_EQUAL if the testbench you want to
     // convert to VUnit already contains code that for example
     // calls $stop if an error-condition is detected.
      end

      `TEST_CASE("Test that a test case that takes too long time fails with a timeout") begin
         $display("This test is expected to timeout because of the watch dog below.");
         #2ns; //
      end

      `TEST_CASE_CLEANUP begin
     // This section will run after the end of a test case. In
     // many cases this section will not be needed.
         $display("Cleaning up after a test case");
      end

      `TEST_SUITE_CLEANUP begin
     // This section will run last before the TEST_SUITE block
     // exits. In many cases this section will not be needed.
         $display("Cleaning up after running the complete test suite");
      end
   end;

   // The watchdog macro is optional, but recommended. If present, it
   // must not be placed inside any initial or always-block.
   `WATCHDOG(1ns);
endmodule`

Python:

`from vunit import VUnit
vu = VUnit.from_argv()
lib = vu.add_library("lib")
lib.add_source_files("../tb/vunit_tb.sv")
vu.main()`

Error:
** Error: ** while parsing macro expansion: 'TEST_SUITE' starting at ..\\vunit_tb.sv(55)

kraigher, 2018-01-15:
It works for me using Modelsim 10.5b. Line 55 is a TEST_CASE in the unmodified example. Try commenting it out? It is strange that Modelsim complains about the TEST_SUITE macro.

alinaivanovaoff, 2018-01-15:
I have commented everything, except
```
`include "vunit_defines.svh"

module vunit_tb;
endmodule
```
I have got 

WARNING - Module vunit_tb matches testbench name regex ^(tb_.*)|(.*_tb)$ but has no parameter runner_cfg and will therefore not be run.
in file D:\Projects\rebuf_cat_mem\vunit_test\tb\vunit_tb.sv
WARNING - Found no test benches using current filter rule:

    Filters entities and modules that have a runner_cfg generic/parameter

    Gives warning when design_unit matches tb_* or *_tb without having a runner_cfg
    Gives warning when a design_unit has  a runner_cfg but the name does not match tb_* or *_tb
    
Compiling into lib: ..\tb\vunit_tb.sv passed
Compile passed
No tests were run!

When my code looks 
```
`include "vunit_defines.svh"

module vunit_tb;
   `TEST_SUITE begin
   end;
endmodule
```
I have got 
** Error: ** while parsing macro expansion: 'TEST_SUITE' starting at .\\vunit_tb.sv(55)
line 55 it is - `TEST_SUITE begin

Do you have any idea what I should comment?

kraigher, 2018-01-15:
Why is it line 55, there must be something above? Also try replacing the TEST_SUITE with the expansion:
```verilog

parameter string runner_cfg = ""; 
import vunit_pkg::*;
initial
  if (__runner__.setup(runner_cfg)) 
    while (__runner__.loop) begin
```

alinaivanovaoff, 2018-01-15:
Above just comments. I think comments should not influence compilation.

alinaivanovaoff, 2018-01-15:
You mean:
```
`include "vunit_defines.svh"
module vunit_tb;
   //`TEST_SUITE begin
    parameter string runner_cfg = ""; 
    import vunit_pkg::*;
    initial
      if (__runner__.setup(runner_cfg)) 
        while (__runner__.loop) begin
          ;
        end
endmodule
```
```
Compiling into lib: ..\tb\vunit_tb.sv failed
=== Command used: ===
C:\modeltech64_10.4\win64\vlog -quiet -modelsimini 
..\vunit_out\modelsim\modelsim.ini -sv -work lib ..\vunit_tb.sv -L vunit_lib -L lib "+incdir+..\Python36\lib\site-packages\vunit\verilog\include"
=== Command output: ===
Compile failed
```

svenka3, 2018-01-15:
Hello @alinaivanovaoff - 

I use Modelsim 10.5c and with your TB & PY code I am able to reproduce your issue. A fix that works for me is:

# from vunit import VUnit
from vunit.verilog import VUnit

Maybe that's what you need?

Regards
Srini

kraigher, 2018-01-15:
@alinaivanovaoff @svenka3 
Just a tip that will make your GitHub posts more readable:
https://help.github.com/articles/creating-and-highlighting-code-blocks/#syntax-highlighting

That is how you quote a code block or command output and get good formatting and syntax highlighting. vhdl and verilog is supported. It also works in a Gitter chat.

svenka3, 2018-01-15:
Thanks @kraigher for that GitHub tip - I have seen it earlier, forgot to use it, but now fixed it :-)

alinaivanovaoff, 2018-01-16:
Thank you @svenka3! It helped. @kraigher I think, I did not have the second part of the message, but now I am not sure. I try to repeat my error, but even without (.verilog) everything is working now.

alinaivanovaoff, 2018-01-16:
@kraigher Thank you for GitHub Tip.

kraigher, 2018-01-16:
@alinaivanovaoff It works without .verilog if you ran once with it corrected since then vunit_pkg was added to the compiled library folder in the vunit_out folder. Just like you would manually vlog a file into a library it stays there until you remove the library. You have to run with `--clean` or remove the vunit_oit folder to clear it.

alinaivanovaoff, 2018-01-16:
Thanks @kraigher. I delete vunit_out and launch without .verilog. So my error message looks:
```
"C:\Program Files\Python36\python.exe" ../sim/run.py
Compiling into lib:       ..\tb\vunit_tb.sv                                                                            failed
=== Command used: ===
C:\modeltech64_10.4\win64\vlog -quiet -modelsimini 
..\sim\vunit_out\modelsim\modelsim.ini -sv -work lib 
..\tb\vunit_tb.sv -L vunit_lib -L lib "+incdir+C:\Program Files\Python36\lib\site-packages\vunit\verilog\include"

=== Command output: ===
** Error: ** while parsing macro expansion: 'TEST_SUITE' starting at ..\tb\vunit_tb.sv(19)
Compile failed
```

kraigher, 2018-01-16:
Maybe Modelsim version you are using did not get an improved error message

alinaivanovaoff, 2018-01-16:
I see @kraigher

closed 2018-01-16 by alinaivanovaoff