VUnit/vunit #298: Active-HDL Lattice Edition

climblinne, 2018-01-15:
Can I use the "Active-HDL Lattice Edition (Version 10.3)" to run VUnit? 
I tried a little bit around and it seems not run properly. 
Is there some experience already available?

LarsAsplund, 2018-01-16:
I'm not sure. We only test with standalone Active-HDL. Can you provide the errors you see?

gsorrenti, 2018-01-16:
I'm also interested in Active-HDL Lattice Edition, but I didn't try yet. I
hope I can do some test this week.

Best regards everybody

LarsAsplund, 2018-01-16:
@gsorrenti Didn't you use VUnit with Lattice before? Was that earlier versions? Experiences?

climblinne, 2018-01-16:
I tested yesterday on two PCs (Win7 & Win10) and I got nearly the same error. Today I started again the Win7-PC and it where running without problems. Then I run the project from yesterday - still not running. It comes the following error:
`
s:\SCM\FPGA\VHDL-PGM>python run.py -v
WARNING - Entity tb_libv matches testbench name regex ^(tb_.*)|(.*_tb)$ but has
no generic runner_cfg and will therefore not be run.
in file s:\SCM\FPGA\VHDL-PGM\libv.vhd
Re-compile not needed

Running test: lib.libv_tb.test_pass
Running test: lib.libv_tb.test_pass2
Running 2 tests

Starting lib.libv_tb.test_pass
Output file: vunit_out\test_output\lib.libv_tb.test_pass_69b21db1c442727f276d0a8
75f0567a15e747701\output.txt

VHDL/Verilog/EDIF/SystemC Simulator build 10.3.3558.6081

(c) 1997-2016 Aldec, Inc. All rights reserved.

License Number 0

VSIMSA: Configuration files: `s:\SCM\FPGA\VHDL-PGM\vunit_out\activehdl\library.c
fg', `C:\lscc\diamond\3.10_x64\active-hdl\BIN\vsimsa.cfg'
 Welcome to VSIMSA!
 This message was printed from `startup.do' macro file.

log "s:\SCM\FPGA\VHDL-PGM\vunit_out\test_output\lib.libv_tb.test_pass_69b21db1c4
42727f276d0a875f0567a15e747701\activehdl\transcript"
log -assert "s:\SCM\FPGA\VHDL-PGM\vunit_out\test_output\lib.libv_tb.test_pass_69
b21db1c442727f276d0a875f0567a15e747701\activehdl\transcript"
source "s:/SCM/FPGA/VHDL-PGM/vunit_out/test_output/lib.libv_tb.test_pass_69b21db
1c442727f276d0a875f0567a15e747701/activehdl/common.tcl"
ELBREAD: Elaboration process.
ELBREAD: Error: Package not found: standard (library: std).
ELBREAD: Error: Elaboration process completed with errors.
VSIM: Error: Simulation initialization failed.
set failed [vunit_load]
if {$failed} {quit -code 1}
fail (P=0 S=0 F=1 T=2) lib.libv_tb.test_pass (0.5 seconds)
`
Then I finally deleted the "vunit_out" folder and then it worked again!

climblinne, 2018-01-16:
What is the official way to run Active-HDL in "Debug"-version, so that I can set breakpoints? Can I recompile from Active-HDL?

LarsAsplund, 2018-01-16:
@climblinne You can always do

``` console
python run.py --clean
```
to get a clean start.

To set the debug compile option you need to use the [set_compile_option](http://vunit.github.io/python_interface.html?highlight=set_compile_option#vunit.ui.VUnit.set_compile_option) method in your run script. End your run script with this

``` python
prj.set_compile_option("activehdl.vcom_flags", ["-dbg"])
prj.main()
```

and then run the script with the -g flag to start debugging.

climblinne, 2018-01-16:
Thanks a lot! Perfect!

closed 2018-01-16 by climblinne