VUnit/vunit #300: Parsing macro

alinaivanovaoff, 2018-01-17:
When I use such macro directly in Modelsim simulation, I don't get any warning. When I use VUnit I get a warning.
Macro:
```systemverilog
`define scsMsg(inf)                  $display("SUCCESS: %s!", inf);
```
Usage of this macro 
```systemverilog
`scsMsg({name, " CHECK DATA: output data is equal to benchmark"})
```
'name' is string

Warning
```
WARNING - Too many arguments got 2 expected 1
at ..\..\common\tb\com_tst_pkg.sv line 348:
                        `scsMsg({name, " CHECK DATA: output data is equal to benchmark"})
```

alinaivanovaoff, 2018-01-17:
If I add additional var:
```systemverilog
string full_name;
full_name = {name, " CHECK DATA: output data is equal to benchmark"};
`scsMsg(full_name)
```
There is no warning.
It's a bit strange.

kraigher, 2018-01-17:
The warning is not printed by Modelsim but by VUnit.  Thus you only see it when running VUnit.

VUnit has a Verilog preprocessor since it it is necessary to perform preprocessing to scan for module dependencies since module can be defined or instantiated in a macro.  You have discovered a bug in the VUnit Verilog preprocessor. It has no consequence for you other than seeing the warning if the macro was not defining an item which effects compile dependencies which it seems your macro does not.

svenka3, 2018-01-17:
Nice explanation. Will you consider adding a unique prefix to your preprocessor messages so that it is easier for users to isolate?

Thanks

kraigher, 2018-01-17:
Sure we could add a prefix so that the user does not get confused. Note though that these warnings occur before any compilation (and output for compiling file ...) started so it could not come from the vendor tool.

kraigher, 2018-01-18:
I made a fix for this now.

kraigher, 2018-01-20:
Now part of 2.4.2 release.

closed 2018-01-20 by kraigher