VUnit/vunit #301: No information about error

alinaivanovaoff, 2018-01-17:
When I launched testbench directly in Modelsim, I got an error message:
```
# ** Error: ../tb/vunit_test_tb.sv(133): (vlog-2730) Undefined variable: 'END_DELAY'.
```
VUnit sends just: 
```
"..\Python36\python.exe" ../sim/run.py
Compiling into lib: ..\tb\vunit_test_tb.sv failed
=== Command used: ===
..\modeltech64_10.4\win64\vlog -quiet -modelsimini 
..\modelsim\modelsim.ini -sv -work lib ..\tb\vunit_test_tb.sv -L vunit_lib -L lib "+incdir+..\Python36\lib\site-packages\vunit\verilog\include" +incdir+../../common/rtl/ +incdir+../../common/tb/

=== Command output: ===

Compile failed
```

kraigher, 2018-01-17:
You are supposed to get the same error on the command line as in the simulator. What you are seeing has never happened to me. Could you provide a complete Verilog file that reproduces the error?

kraigher, 2018-01-17:
Have you tried running the exact same command as VUnit prints in the exact same working directory as you stand when you run the run.py file using Python?

alinaivanovaoff, 2018-01-18:
What is working directory for VUnit?
It happens when the parameter is not defined in the testbench. VUnit doesn't send any error. Just compile failed.
Actually, It happens with all error, syntax issue etc. VUnit doesn't send an error message in the log, but modelsim send. It's very sad. I need to keep two testbench version for modelsim and VUnit.

kraigher, 2018-01-18:
The working directory for compile is the same as where you stand when you run the run.py script.

alinaivanovaoff, 2018-01-18:
I have launch modelsim using VUnit command. 
`C:/modeltech64_10.4/win64/vlog -quiet -modelsimini ../sim/vunit_out/modelsim/modelsim.ini -sv -work lib ../tb/vunit_test_tb.sv -L vunit_lib -L lib "+incdir+C:/Program Files/Python36/lib/site-packages/vunit/verilog/include" +incdir+../../common/rtl/ +incdir+../../common/tb/`

If one parameter doesn't defined in the test I get an error:
```
# ** Error: ../tb/vunit_test_tb.sv(134): (vlog-2730) Undefined variable: 'END_DELAY'.
# 
# child process exited abnormally
```

alinaivanovaoff, 2018-01-18:
Python script sent:
```
Compiling into lib: ..\tb\vunit_test_tb.sv failed
=== Command used: ===
C:\modeltech64_10.4\win64\vlog -quiet -modelsimini ..\sim\vunit_out\modelsim\modelsim.ini -sv -work lib ..\tb\vunit_test_tb.sv -L vunit_lib -L lib "+incdir+C:\Program Files\Python36\lib\site-packages\vunit\verilog\include" +incdir+../../common/rtl/ +incdir+../../common/tb/

=== Command output: ===

Compile failed
```

kraigher, 2018-01-18:
What shell are you using? What happens if you add a "> output.txt" after the command? What is the content of output.txt? Have you tried running VUnit from Windows CMD.exe shell?

alinaivanovaoff, 2018-01-18:
Reference code:
```systemverilog
`timescale 1 ns / 1 ps
//-----------------------------------------------------------------------------
`include "com_tst_mcr.svh"
`include "vunit_defines.svh"
//-----------------------------------------------------------------------------
module vunit_test_tb import vunit_test_param_pkg::*, vunit_test_test_tst_pkg::*, com_tst_pkg::*;  ();
    parameter CLOCK                          = 12.5;
    parameter realtime TIMEOUT               = 1ms;
    parameter RESET_DELAY                    = 100;
    //parameter END_DELAY                      = 1000;
// Stress reset
    parameter RESET_PERIOD_MIN               = 1000;
    parameter RESET_PERIOD_MAX               = 10000;
    parameter RESET_TIME_MIN                 = 100;
    parameter RESET_TIME_MAX                 = 1000;
//-----------------------------------------------------------------------------
// Interfaces
    var clk;
    sys_intf usys                            (.clk(clk));
    data_intf #(
        .DATA_WIDTH                          (DATA_WIDTH)
    ) udata (.clk(clk), .rstn(usys.rstn));

    int test_fail                            = 0;
    int check_fail                           = 0;

//-----------------------------------------------------------------------------
    test #(
        .TIMEOUT                    (TIMEOUT),
        .CLOCK                      (CLOCK),
        .END_DELAY                  (END_DELAY),
        .RESET_DELAY                (RESET_DELAY),
        .RESET_PERIOD_MIN           (RESET_PERIOD_MIN),
        .RESET_PERIOD_MAX           (RESET_PERIOD_MAX),
        .RESET_TIME_MIN             (RESET_TIME_MIN),
        .RESET_TIME_MAX             (RESET_TIME_MAX)
    ) Test;
//-----------------------------------------------------------------------------
// DUT
    vunit_test vunit_test (
    //System intf
        .clk                                 (usys.clk),
        .rstn                                (usys.rstn),
        .data_in                             (udata.data_in),
        .data_out                            (udata.data_out)
    );
//-----------------------------------------------------------------------------
// Clock
    initial begin
        clk = 1;
        forever begin
            #(CLOCK / 2.0) clk               = ~clk;
        end
    end

    `TEST_SUITE begin
        `TEST_CASE("Test_pass") begin
            Test = new(usys, udata);
            Test.pre_test();
            fork
                Test.run();
                Test.test_timeout(usys.clk, test_fail);
            join_any
            @(posedge usys.clk);
            disable fork;
            `CHECK_EQUAL(test_fail, 0);
        end
    end;
    //`WATCHDOG(1us);
endmodule
```
I just comment this line `parameter END_DELAY                      = 1000;`

kraigher, 2018-01-18:
Could you try VUnit in CMD.exe or git bash?

alinaivanovaoff, 2018-01-18:
cmd:
```
Traceback (most recent call last):
  File "D:\Projects\rebuf_cat_mem\vunit_test\sim\run.py", line 4, in <module>
    vu = VUnit.from_argv()
  File "C:\Program Files\Python36\lib\site-packages\vunit\ui.py", line 275, in f
rom_argv
    return cls.from_args(args, compile_builtins=compile_builtins)
  File "C:\Program Files\Python36\lib\site-packages\vunit\ui.py", line 290, in f
rom_args
    return cls(args, compile_builtins=compile_builtins)
  File "C:\Program Files\Python36\lib\site-packages\vunit\ui.py", line 315, in _
_init__
    self._create_output_path(args.clean)
  File "C:\Program Files\Python36\lib\site-packages\vunit\ui.py", line 815, in _
create_output_path
    os.makedirs(self._output_path)
  File "C:\Program Files\Python36\lib\os.py", line 220, in makedirs
    mkdir(name, mode)
PermissionError: [WinError 5] Access is denied: 'C:\\Program Files\\Python36\\vu
nit_out'
```

kraigher, 2018-01-18:
Are you running VUnit from within Python install folder? It tries to create an vunit_out folder in the same direction as you stand but does not get permission. Stand in a folder where you have write  permissions or use --output-path argument.

kraigher, 2018-01-18:
After googling a bit I think PyCharm is the problem https://stackoverflow.com/questions/26465221/debugging-popen-subprocesses-with-pycharm it seems to be modifying any Python process started within it. That probably causes the behavior you observe.

kraigher, 2018-01-18:
On Windows I personslly use git/msys bash via http://cmder.net and can recommend it. Standard CMD.exe should also work.

alinaivanovaoff, 2018-01-18:
Yes, thank you! CMD:
```
D:\Projects\rebuf_cat_mem\vunit_test\sim>python.exe run.py
Compiling into lib: ..\tb\vunit_test_tb.sv failed
=== Command used: ===
C:\modeltech64_10.4\win64\vlog -quiet -modelsimini D:\Projects\rebuf_cat_mem\vun
it_test\sim\vunit_out\modelsim\modelsim.ini -sv -work lib D:\Projects\rebuf_cat_
mem\vunit_test\tb\vunit_test_tb.sv -L vunit_lib -L lib "+incdir+C:\Program Files
\Python36\lib\site-packages\vunit\verilog\include" +incdir+../../common/rtl/ +in
cdir+../../common/tb/

=== Command output: ===
** Error: D:\Projects\rebuf_cat_mem\vunit_test\tb\vunit_test_tb.sv(134): (vlog-2
730) Undefined variable: 'END_DELAY'.
```

kraigher, 2018-01-20:
Since PyCharm is not free software and I do not feel like buying a license for it you would have to try to find the problem in PyCharm yourself and make a pull request for it. It will probably require some Python expertise.

closed 2018-01-20 by kraigher