VUnit/vunit #302: Variable driven in an alwa ys_ff block, may not be driven by any other process

alinaivanovaoff, 2018-01-19:
I have an error from VUnit and have not error from Modelsim:
```
 Error (suppressible): (vopt-7061) ..\rtl\ram.sv(89): Variable 'ram' driven in an alwa
ys_ff block, may not be driven by any other process. See ..\rtl\ram.sv(27).
```
Code 
```systemverilog
module bram # (
    parameter WIDTH  = 16,
    parameter DEPTH  = 64,
    parameter ADBITS = $clog2(DEPTH),
    parameter IS_READ_BUF = 1,
    parameter INIT_FILE = ""
) (
    input  logic              clk,
    input  logic              rstn,
    input  logic              en,
    input  logic              we,
    input  logic [ADBITS-1:0] addr,
    input  logic [WIDTH-1:0]  din,
    output logic [WIDTH-1:0]  dout
);
    (*ram_style = "block" *) logic [WIDTH-1:0] ram [DEPTH-1:0];
    reg [WIDTH-1:0]          ram_data = {WIDTH{1'b0}};

    generate
        if (INIT_FILE != "") begin: use_init_file
            initial begin
                $readmemb(INIT_FILE, ram, 0, DEPTH-1);
            end
        end else begin: init_bram_to_zero
            initial begin
                for (int i = 0; i < DEPTH; i++) begin
                    ram[i] = {(WIDTH){1'b0}};
                end
            end
        end
  endgenerate

    always_ff @(posedge clk) begin
        if (en) begin
            if (we) begin
                ram[addr] <= din;
            end else begin
                ram_data  <= ram[addr];
            end
        end
    end
```

kraigher, 2018-01-19:
This is most likely some missing tool option that you set in your old compile script but have not moved over to a VUnit compile- or sim-option.

alinaivanovaoff, 2018-01-19:
Could you explain more, please?

kraigher, 2018-01-19:
The only possible difference related to your problem is the flags used for the vlog command. What flags did you have originally?

kraigher, 2018-01-19:
Or possibly flag to vsim/vopt

alinaivanovaoff, 2018-01-19:
Do you mean this -sv -lint ?

kraigher, 2018-01-19:
Yes those are flags to the tool. The -sv flag is added by VUnit if the file ends with .sv. The lint flag needs to be added by you using the "modelsim.vlog_flags" compile option. Search for compile_option and simultation_option in our documentation on the website http://vunit.github.io

alinaivanovaoff, 2018-01-19:
Thank you, I fixed this issue. I have added ` lib.set_sim_option("modelsim.vsim_flags", ["-novopt"])` and 
`lib.add_source_files("../tb/module.sv", include_dirs=INCLUDE_DIRS).set_compile_option("modelsim.vlog_flags", ["-lint"])`

closed 2018-01-19 by alinaivanovaoff