VUnit/vunit #303: Modelsim: add wave issue

alinaivanovaoff, 2018-01-19:
I launch Modelsim in GUI mode and add my wave.do file.  But I get a message from Modelsim:
```
# Sourcing file ../sim/wave.do from modelsim.init_files.after_load
# Sourcing ../wave.do failed
# (vish-4014) No objects found matching '../ram_rd_en_r'.
```
It can happen with different signal types and they definitely defined in my design or testbench. I have checked it in Modelsim directly.

kraigher, 2018-01-19:
Try launching the test case with --gui flag and run the wave.do file manually. You can see the list of objects in the gui. There is probably some setting you have on you old scrips that you have not moved over to a compile or simulation option for Modelsim in VUnit.

alinaivanovaoff, 2018-01-19:
Thank you, I fixed this issue. I have added ` lib.set_sim_option("modelsim.vsim_flags", ["-novopt"])` and 
`lib.add_source_files("../tb/module.sv", include_dirs=INCLUDE_DIRS).set_compile_option("modelsim.vlog_flags", ["-lint"])`

closed 2018-01-19 by alinaivanovaoff

sajjadkhan60, 2020-07-26:
> Thank you, I fixed this issue. I have added ` lib.set_sim_option("modelsim.vsim_flags", ["-novopt"])` and
> `lib.add_source_files("../tb/module.sv", include_dirs=INCLUDE_DIRS).set_compile_option("modelsim.vlog_flags", ["-lint"])`

I am also having this problem.
Can you please help me with this ?

sajjadkhan60, 2020-07-26:
How you solved it,?
Please guide me.
thanks

eine, 2020-07-28:
@sajjadkhan60 what you quoted was the solution (I assume, adding `-novopt` and `-int` to `vsim` and `vlog`, respectively), not the problem. Can you please provide more info about your use case and the specific error you get?