VUnit/vunit #304: Post check function launched before simulation

alinaivanovaoff, 2018-01-20:
My part of run.py code:
```python
vdfb_tb = lib.test_bench("vdfb_tb")
vdfb_tb.add_config(post_check=check.post_check(output_path))
vu.main()
```
Problem: my post check function have launched before siulation.

kraigher, 2018-01-20:
You do not pass the function itself but the return value of you calling the function yourself.

alinaivanovaoff, 2018-01-20:
What is the correct way to set my check.post_check function in the post_check hook?

kraigher, 2018-01-20:
Just pass the function without calling it. `function` instead of `function(argument)`.
This is just Python language mechanics and is not specific to VUnit in any way.

kraigher, 2018-01-20:
In general the VUnit documentation does not aim to teach the Python language. Python is one of the world's most popular languages and had plenty of guides, references, tutorials etc available so there is no need for us to provide it. A run.py is just a Python script that uses a Python library named VUnit.

alinaivanovaoff, 2018-01-20:
Yes, I already try this one and I got this error:
```
Traceback (most recent call last):
  File "run.py", line 111, in <module>
    vdfb_tb.add_config(post_check=check.post_check)
TypeError: add_config() missing 1 required positional argument: 'name'
```
That is the the name I should set, if I want to launch post_check after my simulation have finished?

kraigher, 2018-01-20:
add_config creates named configurations of a test bench/test case. It is for the use case where you want to create multiple runs of the same test using different parameters/generics. If you just want to set the post check without creating multiple test runs just use set_post_check.

alinaivanovaoff, 2018-01-20:
Will post_check launch if the test failed?

kraigher, 2018-01-20:
No

alinaivanovaoff, 2018-01-20:
@kraigher In some cases we need launch post_check function even though test failed((( Ok, thanks for the help. I hope all of the numerous VUnit users have good Python skills, but in my company, we are focused on C++ and SystemVerilog. We have found VUnit and it looks not so bad for us, but a few examples not allows to do quick start.

kraigher, 2018-01-20:
When do you need the post_check function to run even if the HDL-level test failed?

kraigher, 2018-01-20:
@alinaivanovaoff I improved the documentation of `pre_config` and `post_check` functions. Let me know if it was what you where looking for.

kraigher, 2018-01-20:
https://vunit.github.io/python_interface.html#pre-and-post-simulation-hooks

alinaivanovaoff, 2018-01-24:
@kraigher Thank you! Sometimes I need launch post_check function, even though my simulation is failed because post_check function performs more details data analysis and it can help to find a bug. For example, if the bug has happened before my simulation is failed.

closed 2018-01-24 by kraigher

alinaivanovaoff, 2018-01-25:
@kraigher Could you add a flag, which indicates closing script if the test fails or not, please?

kraigher, 2018-01-25:
Create a new issue where you request a new feature and describe your requirements in more detail.

alinaivanovaoff, 2018-01-25:
Ok, thank you.