VUnit/vunit #307: Test case name doesn't support string param

alinaivanovaoff, 2018-01-25:
I try to use string localparam in the test case name:
```systemverilog
localparam string NAME = "NAME";
...
`TEST_CASE(NAME) begin
    test_fail = 0
    `CHECK_EQUAL(test_fail, 0)
end
```
I got an error: `# ** Error: Found no "" test case`

kraigher, 2018-01-25:
This is not intended to work. Why would you need it?

alinaivanovaoff, 2018-01-25:
I want to use the same parameter inside my test class and as the name of the test case.

kraigher, 2018-01-25:
Do you instantiate the test class within the TEST_CASE block? In that case we might be able to add a current_test_case which returns a string. 

I do not want any dynamic code in the test case definition itself since we by design want to statically know all test case names without elaborating or running Verilog code.

alinaivanovaoff, 2018-01-25:
Yes, I instantiate test class inside TEST_CASE block.

alinaivanovaoff, 2018-01-25:
> design want to statically know all test case names without elaborating or running Verilog code.

Why is it good? What are advantages?

kraigher, 2018-01-25:
Elaborating or running Verilog code which would be unnecessarily complexity which is always something to avoid. Maybe in your simple case a static check could suffice but in the general case it would not. It is a desirable property of a test system that all test names are known prior to executing the actual code such that listing all test cases using for example `--list` argument does not require running any code to "discover" what where the tests.

alinaivanovaoff, 2018-01-25:
Modelsim allows setting parameters from do file. For example, I can define a list of tests in run.py file and pass it to testbench and to VUnit. It will be defined outside SystemVerilog code. Does it solve this problem?

alinaivanovaoff, 2018-01-25:
It would be good if your documentation mentions about the restriction in the naming TEST_CASE.