VUnit/vunit #308: Launch Vunit tests in the loop

alinaivanovaoff, 2018-02-02:
Is it possible to launch Vunit tests in the loop? I mean, if tests finished successfully - just launch again, if some of the tests failed - exit?

sthenc, 2018-02-02:
If it is the same testcases, why would you do that?

alinaivanovaoff, 2018-02-02:
No, it’s not the same tests. Every time will load new configuration. In the python script will generate new set of parameters and launch VUnit tests.

alinaivanovaoff, 2018-02-02:
Also, it allows to test with different random seeds. So It allows to achieve good code coverage.

kraigher, 2018-02-02:
@alinaivanovaoff Seems like your use case is covered by adding many test configurations using the add_config method. Example: http://vunit.github.io/python_interface.html#example
It allow creating named runs of the same test bench using different configuration of simulator flags, generics/parameters, post_check/pre_config etc.

alinaivanovaoff, 2018-02-03:
Unfortunately it’s not suitable for my case, because I have so much configuration and we don’t plane generate it. If I can launch VUnit in the loop I can combine VUnit run.py script with my others and l can lunch so much configurations or random seed as I want.

kraigher, 2018-02-03:
Is the loop infinite? If you want to run 1000 iterations you can call add_config 1000 times. Also you can always run the run.py file itself in a loop.  Or you can loop at RTL-level instead. There are many options.

alinaivanovaoff, 2018-02-03:
Ok, I’ll try

closed 2018-03-01 by kraigher