VUnit/vunit #310: TLAST not required on every axi_stream_slave transaction

jklapel, 2018-02-21:
Currently, every transaction of the axi_stream_slave requires that TLAST be set to '1', or else a check in the axi_stream_slave entity will fail, causing a failure of the VUnit test. However, this is TLAST is not a requirement of the AXIS protocol and should only come on the last data beat of a packetized data stream.

As a workaround, I commented out the check in axi_stream_slave, as well as added a pop_axi_stream procedure (similar to the push_axi_stream procedure) that returns TLAST so that I can check/act on it myself.

kraigher, 2018-02-22:
So we could add a last-concept to the generic stream master/slave VCI. 

Making the following changes to stream_master:
```vhdl
  -- NOTE: Added last parameter
  -- Push a data value to the stream
  procedure push_stream(signal net : inout network_t;
                        stream : stream_master_t;
                        data : std_logic_vector;
                        last : boolean := false);
```

Making the following additions/changes to stream_slave:
```vhdl
  -- NOTE: New procedure in addition to old
  procedure pop_stream(signal net : inout network_t;
                       stream : stream_slave_t;
                       variable data : out std_logic_vector;
                       variable last : out boolean);

  -- NOTE: New procedure in addition to old
  procedure await_pop_stream_reply(signal net : inout network_t;
                                   variable reference : inout stream_reference_t;
                                   variable data : out std_logic_vector;
                                   variable last : out boolean);

  -- NOTE: Added last parameter
  -- Blocking: read stream and check result against expected value
  procedure check_stream(signal net : inout network_t;
                         stream : stream_slave_t;
                         expected : std_logic_vector;
                         last : boolean := false;
                         msg : string := "");
```

jklapel, 2018-02-22:
That would work for my immediate needs.  (Having the rest of the AXI Stream signals would be useful as well, but that's likely a separate issue.)

Would this make the current push_axi_stream redundant then?

kraigher, 2018-02-23:
@jklapel Yes it would make the AXI-stream specific push-method redundant until we want to add AXI specific flags to the push/pop commands such as tstrb or tkeep or tuser.

kraigher, 2018-02-28:
This is a good first issue if you want to make a pull request to VUnit.

jklapel, 2018-03-06:
@kraigher I created a PR that adds in the suggestions you mentioned above.  I'd appreciate any feedback you have.

jklapel, 2018-03-07:
This has been resolved with PR #314

closed 2018-03-07 by jklapel