VUnit/vunit #311: message passing Pop from empty queue

slaweksiluk, 2018-02-27:
Hello! I'm working on wishbone VC at wishbone branch on my fork.  I encountered a problem which in my opinion might be related to vunit internal message passing system. When I run simple, manual wishbone slave test bench [wishbone slave test bench](https://github.com/slaweksiluk/vunit/blob/wishbone/vunit/vhdl/verification_components/test/tb_wishbone_slave.vhd) it works correctly - 3 writes and then 3 reads are performed. Please note that wishbone_slave.vhd entity uses communication package only internally.

In [tb_wisbhone_master.vhd](https://github.com/slaweksiluk/vunit/blob/wishbone/vunit/vhdl/verification_components/test/tb_wishbone_master.vhd) test I connect wishbone salve and master and test them together. The above test is not passed. Trace is:
```
         5000000 fs - tb                       -    INFO - Writing...
         5000000 fs - vunit_lib:com            -   TRACE - [1:- - ->  (write bus)] =>  inbox
         5000000 fs - vunit_lib:com            -   TRACE -  inbox => [1:- - ->  (write bus)]
         5000000 fs - vunit_lib:com            -   TRACE - [2:- - ->  (write bus)] =>  inbox
         5000000 fs - vunit_lib:com            -   TRACE - [3:- - ->  (write bus)] =>  inbox
         5000000 fs - tb                       -    INFO - Sleeping...
        15000000 fs - vunit_lib:com            -   TRACE -  inbox => [2:- - ->  (write bus)]
        15000000 fs - vunit_lib:com            -   TRACE - [4:- - -> slave ack actor (wb slave write)] => slave ack actor inbox
        15000000 fs - vunit_lib:com            -   TRACE - slave ack actor inbox => [4:- - -> slave ack actor (wb slave write)]
        25000000 fs - vunit_lib:com            -   TRACE - [5:- - -> slave ack actor (wb slave write)] => slave ack actor inbox
        25000000 fs - vunit_lib:com            -   TRACE -  inbox => [3:- - ->  (write bus)]
        25000000 fs - vunit_lib:com            -   TRACE - [6:- - ->  (wb master ack msg)] =>  inbox
        25000000 fs - vunit_lib:com            -   TRACE - slave ack actor inbox => [5:- - -> slave ack actor (wb slave write)]
        35000000 fs - vunit_lib:com            -   TRACE - [7:- - -> slave ack actor (wb slave write)] => slave ack actor inbox
        35000000 fs - vunit_lib:com            -   TRACE - [8:- - ->  (wb master ack msg)] =>  inbox
        35000000 fs - vunit_lib:com            -   TRACE -  inbox => [6:- - ->  (wb master ack msg)]
        35000000 fs - vunit_lib:com            -   TRACE -  inbox => [8:- - ->  (wb master ack msg)]
        35000000 fs - vunit_lib:com            -   TRACE - slave ack actor inbox => [7:- - -> slave ack actor (wb slave write)]
        45000000 fs - vunit_lib:com            -   TRACE - [9:- - ->  (wb master ack msg)] =>  inbox
        45000000 fs - vunit_lib:com            -   TRACE -  inbox => [9:- - ->  (wb master ack msg)]
        45000000 fs - master                   -    INFO - finished wb cycle
       155000000 fs - tb                       -    INFO - Reading...
       155000000 fs - vunit_lib:com            -   TRACE - [10:- - ->  (read bus)] =>  inbox
       155000000 fs - vunit_lib:com            -   TRACE - [11:- - ->  (read bus)] =>  inbox
       155000000 fs - vunit_lib:com            -   TRACE - [12:- - ->  (read bus)] =>  inbox
       155000000 fs - vunit_lib:com            -   TRACE -  inbox => [10:- - ->  (read bus)]
       165000000 fs - vunit_lib:com            -   TRACE -  inbox => [11:- - ->  (read bus)]
       165000000 fs - slave                    -    INFO - push_integer
       165000000 fs - slave                    -    INFO - send
       165000000 fs - vunit_lib:com            -   TRACE - [13:- - -> slave ack actor (wb slave read)] => slave ack actor inbox
       165000000 fs - vunit_lib:com            -   TRACE - slave ack actor inbox => [13:- - -> slave ack actor (wb slave read)]
       165000000 fs - slave                    -    INFO - pop_integer
       165000000 fs - slave                    -    INFO - read word
       175000000 fs - slave                    -    INFO - push_integer
       175000000 fs - slave                    -    INFO - send
       175000000 fs - vunit_lib:com            -   TRACE - [14:- - -> slave ack actor (wb slave read)] => slave ack actor inbox
       175000000 fs - vunit_lib:com            -   TRACE -  inbox => [12:- - ->  (read bus)]
       175000000 fs - vunit_lib:com            -   TRACE - [15:10  -> - (-)] =>  outbox
       175000000 fs - vunit_lib:com            -   TRACE - [16:- - ->  (wb master ack msg)] =>  inbox
       175000000 fs - vunit_lib:com            -   TRACE - slave ack actor inbox => [14:- - -> slave ack actor (wb slave read)]
       175000000 fs - slave                    -    INFO - pop_integer
/home/lab-buddy/git/github/vunit/vunit/vhdl/data_types/src/queue_pkg-body.vhd:104:5:@175ns:(assertion error): Pop from empty queue

```

The code which triggers error is line 99 in wishbone_slave.vhd.
```
			addr := pop_integer(request_msg);
```
Now, because wishbone_slave.vhd is working on standalone run and it doesn't have external communication interface. I suspect it might be vunit internal bug. It is also strange as the integer for that message is pushed in line 69.

Sorry for code formatting and please let me knot if you need any additional information. I'm also think about generating minimal reproducer however it might not be easy. I just merged main repo master branch into my fork wishbone branch.

LarsAsplund, 2018-02-27:
@slaweksiluk The `receive` call in wishbone_master will also delete the message data for `request_msg` used in the previous iteration. This is a convenience when you have a simple "single threaded" BFM. In your case the message is placed in a queue and its data shouldn't be deleted until it has been used on the other side of the queue. You can replace `receive` with the following and then delete the message manually when it's no longer needed.

``` vhdl
      -- receive(net, bus_handle.p_actor, request_msg);
      wait_for_message(net, bus_handle.p_actor, status);
      get_message(net, bus_handle.p_actor, request_msg);
```

where status is of `com_status_t`. You can ignore the status. It's only useful to determine a timeout if you add a timeout parameter to the `get_message` call

LarsAsplund, 2018-02-27:
@slaweksiluk The `receive` call in wishbone_master will also delete the message for `request_msg` used in the previous iteration. This is a convenience when you have a simple "single threaded" BFM. In your case the message is placed in a queue and it shouldn't be deleted until it has been used on the other side of the queue. You can replace `receive` with the following and then delete the message manually when it's no longer needed.

``` vhdl
      -- receive(net, bus_handle.p_actor, request_msg);
      wait_for_message(net, bus_handle.p_actor, status);
      get_message(net, bus_handle.p_actor, request_msg);
```

where status is of `com_status_t`. You can ignore the status. It's only useful to determine a timeout if you add a timeout parameter to the `get_message` call

kraigher, 2018-02-27:
The receive method also deletes the message that is inputed if it is non-null. If you want to keep having ownership of the message across a receive call you can set it to null_message before the call to receive. Semantically the best thing is to set the message to null_message right after pushing it to a queue since you are effectively transferring ownership of the message into the queue.

slaweksiluk, 2018-02-27:
Ok so just to be sure, the problem is in wishbone_master.vhd. Message data is deleted before it is received at the other side of queue. I can fix it as you mentioned, but maybe such a design is non efficient? I've recently seen commit regarding message forwarding. Will it prevent from deleting message?

kraigher, 2018-02-27:
Yes message data is deleted if it is passed to receive method again. You should set the message variable to null after pushing it to the queue since ownership is transferred there. It should not be inefficient. You only delay the deletion of the message. Also only a reference to the message is written to the queue and not the entire message by value.

LarsAsplund, 2018-02-27:
You can also avoid the receive with the code below and then delete the message manually when you're ready for that 

``` vhdl
      -- receive(net, bus_handle.p_actor, request_msg);
      wait_for_message(net, bus_handle.p_actor, status);
      get_message(net, bus_handle.p_actor, request_msg);
```

Status is of type `com_status_t` but can be ignored. It's only useful to determine a timeout if you add a timeout parameter to the `get_message` call.

The forwarding just committed doesn't hand over ownership. It just enables you to push a message into another and doesn't set the input to null. This is something I'm considering though. One of the main points with message passing is to avoid mistakes with shared data by making sure that there is only one owner at any time.

slaweksiluk, 2018-02-28:
Ok thank you, yours code snippet fix it. One more question, do I properly delete request message [here](https://github.com/slaweksiluk/vunit/blob/wishbone/vunit/vhdl/verification_components/src/wishbone_master.vhd#L112)? 

And some off topic question regarding new trace functionality. When message is passed through write_bus() the log is
``
[1:- - ->  (write bus)] =>  inbox
``
Could it be more verbose? For example sender could be "test bench" while receiver "wishbone master". It would make message tracing even more helpful debugging tool. I can make another issue for this request if you find it relevant.

LarsAsplund, 2018-02-28:
@slaweksiluk You would also have to delete the message in the branches where it isn't pushed to the queue.

We have discussed adding support for monitoring memory usage to more easily detect memory leaks. This is still a todo. The best thing you can do now is to do a million transactions or so and see if you get stack overflow or similar from the simulator.

To get more trace information you should give actors names when using `new_actor`. You can also sign a new message with the sending actor by using the sender parameter to `new_msg`.

kraigher, 2018-03-01:
@slaweksiluk Have you got the help you wanted? In that case we can close the issue.

slaweksiluk, 2018-03-02:
Yes I resolved my problem with deleted message. One question regarding trace functionality for bus_master_pkg.vhd. How can I call new_actor() for it? Should I manually overwrite p_actor in below record?
```
  type bus_master_t is record
    -- These fields are private, do not use directly
    p_actor : actor_t;
    p_data_length : natural;
    p_address_length : natural;
    p_byte_length : natural;
    p_logger : logger_t;
  end record;
```

kraigher, 2018-03-08:
You cannot manually overwrite the p_actor since it is passed as a generic and is a constant propagated from the test bench.

slaweksiluk, 2018-03-10:
So is there any way to get more verbose message than
```
[10:- - ->  (read bus)] =>  inbox
```
?

kraigher, 2018-03-10:
I guess not at the moment. We would have to have an optional parameter for the new_master to give the actor a name. We can certainly add that.

LarsAsplund, 2018-03-19:
@slaweksiluk @kraigher I think such an optional parameter is a good idea but it belongs to a new issue. I will open such an issue and close this one.

closed 2018-03-19 by LarsAsplund