VUnit/vunit #313: ghdl architecture obsolated

slaweksiluk, 2018-03-06:
Hello I'm using following script to run vunit tb:
```
from vunit import VUnit
from subprocess import call

# Make sure hdl lib is up to date
call(["make", "VHDL_STD=08"], cwd="../../")

# Create VUnit instance by parsing command line arguments
vu = VUnit.from_argv()

vu.add_external_library("hdl_lib", "../../")

# Create library 'lib'
lib = vu.add_library("lib")

# Add all files ending in .vhd in current working directory to library
lib.add_source_file("axis_fifo_tb.vhd")

vu.main()
```

Entity axis_fifo.vhd is part of hdl_lib library, while test bench axis_fifo_tb.vhd is not. When I call python run.py for the first time it works. When I modify axis_fifo.vhd it is recompiled in library, however I I got following error from ghdl:
```
architecture "bench" of "axis_fifo_tb" is obsoleted by entity "axis_fifo"
```

Then I need to use --clean to force vunit to recompile axis_fifo_tb.vhd. In fact it can be ghdl issue - it should not be needed to reanalyse axis_fifo_tb.vhd as it was not changed...

kraigher, 2018-03-06:
The use case for external libraries is for rarely changed 3rd party dependencies such as the unisim library that are compiled outside of VUnit and which do not need to be incrementally compiled or have dependency tracked by VUnit. In this situation in general a change of the external library always requires a --clean compile within VUnit since the specific dependencies between the VUnit managed files and the external library files are not know. 

In the specific case such as yours one could argue that a change of the external library should not require a recompile of a VUnit tracked file but that is not generally true and depends on code and simulator specifics. In your case GHDL considers the test bench architecture dependent on the entity definition of axis_fifo and I believe other simulators would agree.

slaweksiluk, 2018-03-07:
Yes it is right to assume that library is rather rarely changed. As a workaround for library in development I simply modify axis_fifo_tb also (e.g. add white character) to force vunit to recompile only it. Then I don't have to apply --clean every time.

closed 2018-03-07 by slaweksiluk