VUnit/vunit #316: Make 'latest release' point to the latest published release

1138-4EB, 2018-03-15:
At now the latest published item in [releases](https://github.com/VUnit/vunit/releases) is 'v3.0.2': . However, https://github.com/VUnit/vunit/releases/latest is redirected to [v0.51.3](https://github.com/VUnit/vunit/releases/tag/v0.51.3). As a result, it is not straightforward to download the latest stable release programatically, i.e., `curl -s https://api.github.com/repos/vunit/vunit/releases/latest | grep "tag_name"`.

I believe this is not a critical issue, because most of the users will install the library through `pip`. Still, it might be worth checking the integration with travis-ci.

kraigher, 2018-03-15:
We stopped using the release feature of github some time ago and instead publish release notes on the web-page. I think the github release logic automatically picks up when we push a tag with vX.Y.Z which we did not indent. Maybe the Travis CI script can be modified to update the latest github release pointer as well.

1138-4EB, 2018-03-15:
Indeed, IMHO you are doing it the other way round. The actual solution is perfectly fine for your needs. However, that is not the 'preferred' workflow to synchronize GitHub releases and travis-ci.

Right now you don't make any difference between a 'release' commit and any other regular commit. Then, you have some hand-written logic to decide when a release is to be built. Last, if a release is pushed, a tag is created in travis-ci and it is pushed to github.

The preferred workflow is to create a tag locally when a release is to be built. As soon as the tag is pushed to GitHub, travis indentifies it as release commit. Then, if all the tests are successful, a tarball is built and it is automatically pushed to GitHub Releases as a new release (that is marked as the latest).

The main advantage of this approach is that the git tree is not modified by travis at all. Tags are manually pushed and what gets automated is the deployment to GitHub releases only.

See:

- https://docs.travis-ci.com/user/deployment/releases/
- https://github.com/ghdl/ghdl/blob/master/.travis.yml#L27-L36

Shall you want to change it, I recommend to:

- Backup `.travis.yml`
- Install travis cli.
- Run `travis setup releases` from the root of the repo and follow the steps.
- Clean up `.travis.yml` and check that nothing was messed up.

I'd like to contribute with a PR, but this needs to be done by a maintainer.

1138-4EB, 2018-03-22:
This part of the contribution guidelines is related to the explanation above: https://github.com/VUnit/vunit/blob/master/docs/contributing.rst#making-releases

kraigher, 2018-03-22:
Yes we will probably do this eventually

PR #508 (2019-06-04): Rework release procedure
Close #316.

This PR is an attempt to rework the release procedure through Travis.

- The job to deploy to GitHub Pages is merged with `py36-docs`. The deployment is only executed with tagged commits.
- Script `tools/is_new_release.py` is renamed to `tools/new_release.py` and it is reworked:
  - In order to tag a commit, the maintainer will execute `./tools/new_release.py` locally and then `git push v<version>` to the main repository. The tag will fail/skip if it already exists, or if corresponding release notes are not found.
  - In travis, `./tools/new_release.py 'ch [... cut]
closed 2019-06-04 by kraigher