VUnit/vunit #318: Register VUnit at 'awesome-vhdl' list.

Paebbels, 2018-04-07:
I created a new [awesome list](https://github.com/sindresorhus/awesome) based on the [awesome-python list](https://github.com/vinta/awesome-python).

It's available here: https://github.com/VHDL/awesome-vhdl

Please register VUnit as a verification framework (or propose another or new category) in that list via pull request.

------------------
I this list ist filled with some projects, I want to register it at the list of lists: [awesome list](https://github.com/sindresorhus/awesome)

dalex78, 2022-05-09:
The linked repository has been archived by the owner. It is now read-only. 
Alternative: [awesome-hardware-tools](https://github.com/aolofsson/awesome-hardware-tools) which already has VUnit listed.

@LarsAsplund, I suggest to close.

closed 2022-05-09 by LarsAsplund

umarcor, 2022-05-09:
See also https://github.com/hdl/awesome and https://hdl.github.io/awesome/items/vunit/.