VUnit/vunit #325: Testers Needed for Cadence Incisive and Xcelium

LarsAsplund, 2018-04-18:
The core development team for VUnit does not have easy access to Cadence Incisive and Xcelium licenses which prevents us from running our acceptance tests on those simulators. To get a higher confidence in the updates we release we need feedback from the VUnit users using those simulators. What we're looking for is feedback in general and people that can run our acceptance tests on new release candidates. Our acceptance tests are part of the VUnit installation and running them is just a matter of starting a test suite from the command line to get a test report.

If you interested in the continuous support of these simulators you can comment on this issue or contact [me](https://github.com/LarsAsplund) directly.

petandpal, 2019-05-22:
Hello, I am trying to run Vunit with Xcelium (using the incisive simulator option) but I am seeing some strange issues. First it cannot find the installation root and is looking for Virtuoso. It then goes forward and gets stuck with vunit_lib not found even though it is specified in the cds.lib file. Any help with this?

setenv VUNIT_SIMULATOR incisive
My path points to /XCELIUM1809/tools/bin

[image]

1138-4EB, 2019-05-22:
@petandpal, I think that this issue is for users willing to contribute by running the acceptance tests of VUnit itself. Hence, it is probably not the best place to repost a specific issue with a custom project that you are trying to run. Maintainers are aware of your other issue, and putting it here is not good netiquette.

LarsAsplund, 2019-05-23:
If you have a Cadence license you can go and support https://support.cadence.com/apex/COSCaseDetailPage?CaseID=5000V00001J93GT

powARman, 2020-03-27:
I am currently using xcelium 19.09 and am willing to help on improving support for it in vunit. Some of the issues with VHDL2008 support (e.g. the compile issue with the decode function) have already been fixed in current hotfix releases of xcelium. The only files that currently cannot be compiled with xcelium are log_deprecated_pkg.vhd and check_deprecated_pkg.vhd, so I just removed them in my installation of vunit. Everything else seems to work fine so far. I made some patches to vunit source to further improve the xcelium support, e.g. the ability to disable IEEE warnings.
I will soon provide some patches via pull requests. One question I have: Shall I include xcelium into the existing incisive.py file or create a new one? When using the current implementation with xcelium I get many warnings regarding deprecated command line parameters so it may be clearer to have a separate implementation for xcelium.

LarsAsplund, 2020-03-27:
@powARman It would be great if you can help us out here. Have you tried running the acceptance tests (http://vunit.github.io/contributing.html#running-the-tests)?. I think we should split Incisive from Xcelium if there are warnings in the way you describe. I guess the two could share a lot of code though.

powARman, 2020-04-22:
@LarsAsplund I tried running the tests, at first most failed. After fixing a compile issue with Xcelium and enabling VHDL context support (finally supported by Xcelium) it looks much better. I will further investigate this.
Regarding adding support for Xcelium into vunit: The differences between Incisive and Xcelium are not that big, so I would still support them in one simulator interface class. I am considering renaming incisive.py to a more generic cadence.py. Would that be OK?

vperrin59, 2020-09-14:
I am currently using xcelium and using verilog / SystemVerilog. I'm also willing to help on improving support for it in Vunit.

LarsAsplund, 2020-09-15:
@vperrin59 We would appreciate that. The first step would be to run the entire [test suite](http://vunit.github.io/contributing.html#running-the-tests) to understand what works today and what doesn't work. Is that something you can start with?

vperrin59, 2020-09-15:
@LarsAsplund I'm still trying to get used to the vunit architecture and building some benches, but I will try running the test suite and check the results

rodrigomelo9, 2021-08-29:
Hi. I have access (work) to a Centos 7 with Xcelium 19.09-s002. I have not admin rights, so I installed pyenv, python 3.7.0 and I tried `pytest tests/acceptance`, which never ends (I run `python3 tools/incisive_vhdl_fixup.py` previously):

```
latform linux -- Python 3.7.0, pytest-6.2.4, py-1.10.0, pluggy-0.13.1
rootdir: /<PATH>/vunit
collected 49 items                                                                                                                                                                                        

tests/acceptance/test_artificial.py FFssFsF                                                                                                                       [ 14%]
tests/acceptance/test_dependencies.py F                                                                                                                         [ 16%]
tests/acceptance/test_external_run_scripts.py FFFFFFFFFFFFFFFFFFFFFFFF
```

Here you have the full output: [run.log](https://github.com/VUnit/vunit/files/7071792/run.log) (I removed path names).

Take into account that I am starting with Vunit, so maybe I did something wrong (but I tried in my PC with ghdl and it worked).

I am aware of #504 #677 #684 

Please, let me know how can I help, what to test, check, etc.

Rodrigo

rodrigomelo9, 2021-08-29:
I also tried `examples/v*/user_guide`:

# VHDL

```
$ python3 run.py 
Compiling into vunit_lib: ../../../vunit/vhdl/string_ops/src/string_ops.vhd                        passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
Compiling into vunit_lib: ../../../vunit/vhdl/path/src/path.vhd                                    passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
Compiling into vunit_lib: ../../../vunit/vhdl/logging/src/print_pkg.vhd                            passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
Compiling into vunit_lib: ../../../vunit/vhdl/data_types/src/types.vhd                             passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
Compiling into vunit_lib: ../../../vunit/vhdl/data_types/src/codec_builder.vhd                     passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
Compiling into vunit_lib: ../../../vunit/vhdl/data_types/src/codec_builder-2008p.vhd               passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
Compiling into vunit_lib: ../../../vunit/vhdl/data_types/src/codec.vhd                             passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
Compiling into vunit_lib: ../../../vunit/vhdl/data_types/src/codec-2008p.vhd                       passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
Compiling into vunit_lib: ../../../vunit/vhdl/data_types/src/api/external_string_pkg.vhd           passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
  impure function read_char (
                          |
xmvhdl_p: *W,NORETN (/<PATH>/vunit/vunit/vhdl/data_types/src/api/external_string_pkg.vhd,35|26): No return statement in the body of function READ_CHAR.
  impure function get_ptr (
                        |
xmvhdl_p: *W,NORETN (/<PATH>/vunit/vunit/vhdl/data_types/src/api/external_string_pkg.vhd,42|24): No return statement in the body of function GET_PTR.
Compiling into vunit_lib: ../../../vunit/vhdl/data_types/src/string_ptr_pkg.vhd                    failed
=== Command used: ===
/<TOOL>/19/19.09.002/tools.lnx86/bin/irun -f /<PATH>/vunit/examples/vhdl/user_guide/vunit_out/incisive/irun_compile_vhdl_file_vunit_lib.args

=== Command output: ===
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
  alias decode_ptr_t is decode[string return ptr_t];
                   |
xmvhdl_p: *E,MMSIG (/<PATH>/vunit/vunit/vhdl/data_types/src/string_ptr_pkg.vhd,115|19): signature mismatch. 93[4.3.3].
xrun: *E,VHLERR: Error during parsing VHDL file (status 1), exiting.

Compile failed
```

# Verilog

```
Compiling into vunit_lib: ../../../vunit/verilog/vunit_pkg.sv passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
xmvlog: *W,NOTIND: unable to access -INCDIR b'/<TOOL>/19/19.09.002'/tools/spectre/etc/ahdl/ (No such file or directory).
package vunit_pkg;
                |
xmvlog: *W,TSNSPK (/<PATH>/vunit/vunit/verilog/vunit_pkg.sv,7|16): `timescale is not specified for the package.  The default timescale of 1ns/1ns will be assumed for this package.
Compiling into lib:       tb_example_basic.sv                 passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
xmvlog: *W,NOTIND: unable to access -INCDIR b'/<TOOL>/19/19.09.002'/tools/spectre/etc/ahdl/ (No such file or directory).
Compiling into lib:       tb_example.sv                       passed
xrun: *W,OPDEPRREN: Command Line Option (-nclibdirname) is deprecated. Use (-xmlibdirname) instead.
xmvlog: *W,NOTIND: unable to access -INCDIR b'/<TOOL>/19/19.09.002'/tools/spectre/etc/ahdl/ (No such file or directory).
Compile passed

Starting lib.tb_example_basic.all
Output file: /<PATH>/vunit/examples/verilog/user_guide/vunit_out/test_output/lib.tb_example_basic.all_58c2666aec61dbf9891244ed2152cee42a949488/output.txt
^C
==== Summary ====================================
skip lib.tb_example_basic.all (102.0 seconds)
=================================================
pass 0 of 1
skip 1 of 1
=================================================
Total time was 102.0 seconds
Elapsed time was 102.1 seconds
=================================================
Some skipped!
WARNING: Test execution aborted after running 1 out of 4 tests
```

A `ctrl-c` was needed.

LarsAsplund, 2021-08-29:
@rodrigomelo9 It would be great if you can help out with this! Let's start with the basics and then move up. Could you start a GitHub fork and then begin like this
1. Get rid of that annoying warning. `nclibdirname` is used in a number of places in incisive.py
2. Get the simple user guide examples working.
    * For the VHDL compile error I would try moving that alias declaration from line 115 to line 107 so it sits right after the decode function it applies to. Maybe it gets confused by the later decode function.
    * For the Verilog test I guess that single stepping through the code in the GUI to figure out where it gets stuck is a good start

cmarqu, 2021-08-29:
I was under the impression that running the fixup script would not be needed with such recent Xcelium versions, but I may be wrong.
I cannot look at the log right now; is the -v200x option used? If not, try adding it.

rodrigomelo9, 2021-08-29:
Ok @LarsAsplund I already created a DRAFT PR #728  to track what I am doing (I guess that I will need to ask and report several things). I started with a copy of incisive.py, called xcelium.py, where Incisive and run were replaced by Xcelium and xrun.

I didn't use the -v200x option @cmarqu, I will try. Ok about the fixup script probably not needed.

rodrigomelo9, 2021-08-30:
> * For the VHDL compile error I would try moving that alias declaration from line 115 to line 107 so it sits right after the decode function it applies to. Maybe it gets confused by the later decode function.

@LarsAsplund I tried that and it works... It was needed to apply in more than one file. Is safe? Can I commit that or it would produce fails with other simulators?

LarsAsplund, 2021-08-30:
@rodrigomelo9 That should be safe.