VUnit/vunit #330: add_source_file : Add argument to disable code coverage for selected file/files

krishnan-gopal, 2018-04-24:
This is particularly very useful for bigger testbenches, where code coverage doesnt need to be collected.  By specifying which files need to be enabled/disabled for code coverage, the simulation performance can be drastically improved.

kraigher, 2018-04-24:
Do you need to selectively control type of coverage for individual files or just on/off with respect to the type of coverage provided by the `--coverage` argument.

krishnan-gopal, 2018-04-24:
It's enough to turn off for a file/ files so that --coverage doesn't have any effect on that file. Basically the result is that the selected file/files are always compiled without coverage while the other files are dependent on the --coverage argument

commit d86c49d6 (2018-04-25): Add disable_coverage compile_option. Closes #330.
closed 2018-04-25 by kraigher