VUnit/vunit #331: add command-line option to clean PASSed testcase data

krishnan-gopal, 2018-04-25:
While dealing with several testcases in a testbench, a regression run could generate huge amounts of data. 
Its only interesting to see whether a testcase result is a PASS or FAIL, and store the generated data (do files, transcript, wavefile) only for the failed testcases.
A command-line option could be added to remove the simulation directory and all the generated files once  a testcase simulation is finished and has a PASS result. The data can of course be saved for the FAIL cases.
This would save a lot of disk space while dealing with huge testbenches and large number of testcases.

kraigher, 2018-04-25:
So you are proposing that the data is cleaned immediately after the test is done?
That is a bit tricky since the output folder of a test case is used to gather coverage database files

kraigher, 2018-04-25:
Has this been a practical problem for you? At work I have 2000 tests and no problem with disk space.

krishnan-gopal, 2018-04-25:
Depends on how much data each simulation needs. Think about testvector/ascii files which may be used for simulations. 
By the way, have you thought about doing the coverage merge after each testcase is finished ? The simulation directory can be deleted after that.
Again, this is more of a convenience feature than a necessity.

kraigher, 2018-04-25:
Maybe merging the coverage file directly would be better. Anyway it adds to the effort needed to implement this. Still it is a worthwhile feature to consider when the time is available.