VUnit/vunit #333: --clean destroy the simulator database in GUI mode

asicnet, 2018-05-17:
Hi

if i call run.py with the option -g --clear
the simulstor is starting the simulation well 

but if i call the gui

vunit_restart 

# Re-compiling using command C:\MyPrograms\Python\python.exe -u ....\ctrl_system\vunit\run.py --compile -g --clean

the simulator is crashing because the --clean option is copied to the common.do file

Please remove this option from the list because teh --clean option removes the simulator database

I remove the sys.arg from the list in vsim_simulator_mixin.py line 60

        recompile_command = [
            sys.executable,
            "-u",
            sys.argv[0],
            "--compile"]  ####  + sys.argv[1:]
        recompile_command_visual = " ".join(recompile_command)

but better to search for --clean and remove this from the list.

---------------------------------------------------
I get always the warning 
** Warning: ......\vunit\vhdl\logging\src\ansi_pkg.vhd(149): Function 'drop_color' may complete without a RETURN at line 148.

maybe it is possible to solve this.

see you
Helmut

kraigher, 2018-05-18:
I fixed to issues for you. This could have been a pull request also since you already had the solution.

closed 2018-05-18 by kraigher

asicnet, 2018-05-18:
I test it. Very helpfull
Thanks
Helmut