VUnit/vunit #337: Issue runner

1138-4EB, 2018-05-24:
This is just a test issue to try a script that should allow to run/test minimum (non)working examples provided in a issue.

```
#>> run.sh
docker run --rm -tv $(pwd):/src -w /src ghdl/ext:vunit python run.py -v

#>> run.py
from os.path import join, dirname
from vunit import VUnit
vu = VUnit.from_argv()
vu.add_library("lib").add_source_file(join(dirname(__file__), "tb_repro.vhd"))
vu.main()

#>> tb_repro.vhd
library vunit_lib;
context vunit_lib.vunit_context;

entity tb_repro is
  generic ( runner_cfg : string );
end entity;

architecture tb of tb_repro is
begin
  main: process
  begin
    test_runner_setup(runner, runner_cfg);
    info("Hello world!");
    test_runner_cleanup(runner);
    wait;
  end process;
end architecture;

#>> end
```

[tb_mwe.vhd.txt](https://github.com/VUnit/vunit/files/2037481/tb_mwe.vhd.txt)

1138-4EB, 2018-05-25:
It works! `curl -L https://raw.githubusercontent.com/1138-4EB/vunit/issue-runner/issue_runner/issue_runner.sh | sh -s 337` will:

- Create a subdir named `issue-337`.
- Save `run.sh`, `tb_repro.vhd` and `tb_mwe.vhd.txt` from the first message of the issue to separate files.
- Execute `run.sh`.

Requirements:

- GitHub does not allow uploading `.vhd` files. So, `.txt` must be appended.
- The script parses the body to search for `#>> filename.ext` lines. All the content between such a token and the next one is saved to the file. Therefore, a las dummy token, `#>> anything` must be added.

Provided feature:

Users can put as many files as desired in the first message of the issue. Any maintainer/collaborator can test provided (non)working example in any machine with docker. It can be used in [play-with-docker.com](https://labs.play-with-docker.com/).

The single limitation is that the docker image that is used in `run.sh` must be available. There is a workaround for issues requiring tools that can not be public: edit `issue_runner.sh` to add a sed command before the execution of `run.sh`.

The sources are just a couple of short cripts: https://github.com/1138-4EB/vunit/tree/issue-runner/issue_runner

LarsAsplund, 2018-05-25:
I think the greatest value in a "playground" is the ability to let non-VUnit users have a test drive. I have no experience with play-with-docker but I had some good discussions with the EDA Playground maintainers earlier this year and hope that they will add support in the near future. Please add a thumbs up to https://github.com/edaplayground/eda-playground/issues/18 to give that issue some extra attention.

1138-4EB, 2018-05-25:
play-with-docker is meant to be a playground for docker and kubernetes. Users are expected to create multiple nodes/containers/machines, connect them, orchestrate the services, etc. That's overkill as a HDL playground, where a single node suffices. However, it is really useful to have a machine with 4GB that lasts for 4h, where you can run "any" distro you want with any tool.

This issue is not about a playground for non-VUnit users, although it can be used as so. The greatest value from my point of view is that VUnit users can report issues, without having to upload the sources that reproduce the issue to any external place. Right now, I need to add them to a repo of my own, to a Gist, etc. Then, maintainers/collaboratos, need to understand how to get them in order to execute and reproduce the example. Sometimes it is straightforward, sometimes minor modifications are required, and sometimes the OP does not provide enough info to reproduce it. This let's the user ensure that the collaborator will get exactly the same results.

---

Anyway, I see that the `issue-runner` scripts can be used not only with VUnit, but with any project in GitHub. Therefore, I am going to split it to it's own repo.

---

I really like EDA Playground as a concept. Indeed, 1138-4EB/hwd-ide is a prototype of a similar solution. However, there are some important limitations right now:

- AFAIK, it is free as in 'free beer'. Not free software. Not even open source.
- *Each run is limited to 60 seconds runtime and 100MB of memory*.
- It is lacking GHDL and VUnit support.
- There is no straightforward mechanism to download the playground (the sources that compose a project) and execute it locally.

Should it be open, we could easily add all the other features and even more. Indeed, should EDA Playground be available as a docker image (including the subset of free tools: iverilog, yosys, GHDL, VUnit, etc.), it'd be straightforward for each user to run it's own EDA playground inside play-with-docker or in their own machines. Thus, overcoming the runtime and memory limitation, without Doules being required to add better servers.

Just as an example, this is a contribution I did some months ago to https://github.com/portainer/portainer There is an 'official' demo in a private server that is available at  http://demo.portainer.io, but it goes down quite frequently. So, I provided [PWD/?stack=portainer-demo/play-with-docker/docker-stack.yml](http://play-with-docker.com/?stack=https://raw.githubusercontent.com/portainer/portainer-demo/master/play-with-docker/docker-stack.yml) as an alternative. It reproduces the 'official' demo in a PWD instance.

1138-4EB, 2018-05-25:
I'm closing this because I moved the scripts to a separate repo. If any user finds this useful, the first comment in this issue is a valid example.

closed 2018-05-25 by 1138-4EB