VUnit/vunit #339: Deceptive Errormessage

doelen, 2018-05-31:
When using vunit and modelsim, the misleading Error message listed below will occur if there is an additional semicolon on the last port definition in an entity. 

VHDL code that generates the error: 
```
entity ReArrangeIN is
  generic(
    g_DataWidth : integer := 32
    );
  port (
    i_DataValid : in  std_logic;
    o_DataValid : out std_logic;
    Clr          : in  std_logic;
    o_counter    : out unsigned(1 downto 0);   -- This last semicolon leads to an IndexError in Python
    );
end ReArrangeIN;
```

```
Traceback (most recent call last):
  File "run.py", line 116, in <module>
    create_test_suite(prj)
  File "run.py", line 91, in create_test_suite
    lib.add_source_files(pattern=sources_to_add)
  File "/usr/local/lib/python3.5/dist-packages/vunit/ui.py", line 1149, in add_source_files
    for file_name in file_names])
  File "/usr/local/lib/python3.5/dist-packages/vunit/ui.py", line 1149, in <listcomp>
    for file_name in file_names])
  File "/usr/local/lib/python3.5/dist-packages/vunit/ui.py", line 1192, in add_source_file
    no_parse=no_parse)
  File "/usr/local/lib/python3.5/dist-packages/vunit/project.py", line 126, in add_source_file
    no_parse=no_parse)
  File "/usr/local/lib/python3.5/dist-packages/vunit/project.py", line 818, in __init__
    design_file = vhdl_parser.parse(self.name)
  File "/usr/local/lib/python3.5/dist-packages/vunit/vhdl_parser.py", line 37, in parse
    database=self._database)
  File "/usr/local/lib/python3.5/dist-packages/vunit/cached.py", line 45, in cached
    result = function(content)
  File "/usr/local/lib/python3.5/dist-packages/vunit/vhdl_parser.py", line 68, in parse
    return cls(entities=list(VHDLEntity.find(code)),
  File "/usr/local/lib/python3.5/dist-packages/vunit/vhdl_parser.py", line 325, in find
    yield VHDLEntity.parse(sub_code[:match.end()])
  File "/usr/local/lib/python3.5/dist-packages/vunit/vhdl_parser.py", line 345, in parse
    ports = cls._find_port_clause(code)
  File "/usr/local/lib/python3.5/dist-packages/vunit/vhdl_parser.py", line 397, in _find_port_clause
    code[match.start(): match.end() + closing_pos + match_semicolon.end()])
  File "/usr/local/lib/python3.5/dist-packages/vunit/vhdl_parser.py", line 474, in _parse_port_clause
    port_list.append(VHDLInterfaceElement.parse(interface_element, is_signal=True))
  File "/usr/local/lib/python3.5/dist-packages/vunit/vhdl_parser.py", line 574, in parse
    mode_split = interface_element_string.split(':')[1].strip().split(None, 1)
IndexError: list index out of range
```

I would expect vunit to respond with a clearer error message such as “unexpected semicolon on line....". It seems that modelsim is not reporting this typo during compilation and the vhdl_parser cannot handle it. 

I'd be happy to help, if you need more information concerning my code and setup.

joshrsmith, 2018-05-31:
This looks like a duplicate of #334. Are you running VUnit >= 3.3.0?

kraigher, 2018-06-01:
#334 fixed the problem that VUnit stopped when there was a parse error using the internal parser inhibiting the real compiler from giving a better error.

The VUnit parser is just regex based and will be worse in the presence of syntax errors in general. VUnit has no real need for a complete parser and developing one takes a lot of time and would be less robust during development. This is why we let the real compiler handle files where the internal parser failed.

Maybe this specific instance of the general problem of the regex based parser being worse can be fixed.

doelen, 2018-06-05:
Yes, that is the same error. Sorry I didn’t see that post.

closed 2018-06-05 by doelen