VUnit/vunit #342: Error with large arrays and GHDL.

qarlosalberto, 2018-06-07:
Hello,

Im having problems with:

```
variable in_mod : integer_array_t;
in_mod := load_csv("./input.csv",g_SIZE_MOD,true);
```

When **input.csv** is a large file GHDL crash without reason :S

Thank you.

1138-4EB, 2018-06-07:
How large is "large"? Did you test the limit size that does no crash?

qarlosalberto, 2018-06-07:
With this file crash:  
[out_maxhold_4096FFT_2paral_12mod_10numFFT.txt](https://github.com/VUnit/vunit/files/2080966/out_maxhold_4096FFT_2paral_12mod_10numFFT.txt)

With this file doesnt crash:

[in_mod_4096FFT_2paral_12mod_10numFFT.txt](https://github.com/VUnit/vunit/files/2080984/in_mod_4096FFT_2paral_12mod_10numFFT.txt)

LarsAsplund, 2018-06-07:
@qarlosalberto No problem on my computer. I'm running GHDL 0.35. What version are you on?

qarlosalberto, 2018-06-07:
```
GHDL 0.36-dev (v0.35-176-gd313f785) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

LarsAsplund, 2018-06-07:
@qarlosalberto Ok, so let's make sure we are running exactly the same thing. This is what I tested

``` vhdl
library vunit_lib;
context vunit_lib.vunit_context;

entity tb_integer_array_crash is
  generic (runner_cfg : string);
end entity;

architecture tb of tb_integer_array_crash is
begin
  main : process
    variable in_mod : integer_array_t;
  begin
    test_runner_setup(runner, runner_cfg);
    in_mod := load_csv("D:/Downloads/out_maxhold_4096FFT_2paral_12mod_10numFFT.txt",32,true);
    info("Length: " & to_string(length(in_mod)));
    test_runner_cleanup(runner);
  end process;
end architecture;
```

qarlosalberto, 2018-06-07:
Yes, the same.

```
cruiz@DASPC034:~/repositorio/98410015_MAXHOLD/tb(1-creacion-core)$ python run.py --ide ise --verbose
Compiling into maxhold_top_tb_lib: maxhold_top_tb.vhd passed
Compile passed

Running test: maxhold_top_tb_lib.tb_integer_array_crash.all
Running 1 tests

Starting maxhold_top_tb_lib.tb_integer_array_crash.all
Output file: vunit_out/test_output/maxhold_top_tb_lib.tb_integer_array_crash.all_b61ddf365c91210aa868dc9212d5a2f39d317540/output.txt
fail (P=0 S=0 F=1 T=1) maxhold_top_tb_lib.tb_integer_array_crash.all (10.8 seconds)

==== Summary =========================================================
fail maxhold_top_tb_lib.tb_integer_array_crash.all (10.8 seconds)
======================================================================
pass 0 of 1
fail 1 of 1
======================================================================
Total time was 10.8 seconds
Elapsed time was 10.9 seconds
======================================================================
Some failed!
```

```
--! Librería estándar.
library ieee;
--! Elementos lógicos.
use ieee.std_logic_1164.all;
--! Funciones aritméticas.
use ieee.numeric_std.all;
--
library std;
use std.textio.all;

-- vunit
library vunit_lib;
context vunit_lib.vunit_context;

use vunit_lib.array_pkg.all;
use vunit_lib.integer_array_pkg.all;

library vunit_lib;
context vunit_lib.vunit_context;

entity tb_integer_array_crash is
  generic (runner_cfg : string);
end entity;

architecture tb of tb_integer_array_crash is
begin
  main : process
    variable in_mod : integer_array_t;
  begin
    test_runner_setup(runner, runner_cfg);
    in_mod := load_csv("/home/cruiz/repositorio/98410015_MAXHOLD/tb/in_mod_4096FFT_2paral_12mod_10numFFT.csv",32,true);
    info("Length: " & to_string(length(in_mod)));
    test_runner_cleanup(runner);
  end process;
end architecture;
```

LarsAsplund, 2018-06-07:
@qarlosalberto Looks like GHDL then but you should probably verify that 0.35s work on you side to make sure that there are no other environmental issues.

qarlosalberto, 2018-06-07:
Ok, im going to try!

LarsAsplund, 2018-06-13:
@qarlosalberto Any news on this one?

qarlosalberto, 2018-06-13:
I will back tomorrow to the work :)

qarlosalberto, 2018-06-18:
I have the same problem. I have done this tests:
- In my PC: Ubuntu 18.04
```
GHDL 0.36-dev (v0.35-176-gd313f785) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
- In a Ubuntu 18.04 Docker:
```
cruiz@dabd098c2a33:~/98410015_MAXHOLD/tb$ ghdl --version
GHDL 0.35rc1 (2017-03-01-551-gdb2a0faa) [Dunoon edition]
 Compiled with GNAT Version: 7.3.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
It can be a RAM problem? I have 8G. I can try in other PC with 16G.

1138-4EB, 2018-06-19:
@LarsAsplund I can reproduce the issue with the latest `ghdl/ext:vunit` image. Rename the attached file to either `tgz` or `tar.gz`, extract it and run `docker run --rm -itv $(pwd):/src -w /src ghdl/ext:vunit python3 run.py`.

[vunit-342.txt](https://github.com/VUnit/vunit/files/2113467/vunit-342.txt)

LarsAsplund, 2018-06-28:
@qarlosalberto I ran with the `--log-level debug` flag and receives the return code -11. Tristan at GHDL suggests that this may be related to ghdl/ghdl#611.

kraigher, 2018-08-16:
Close as GHDL bug?

qarlosalberto, 2019-01-21:
It has been resolved:

https://github.com/ghdl/ghdl/issues/611

1138-4EB, 2019-01-21:
@qarlosalberto did you set the stack size with `ulimit` or did it work as is?

qarlosalberto, 2019-01-28:
mmmm... with ulimit and my GHDL version doesn't work. I am going to try with the new GHDL version.

std-max, 2022-05-31:
@umarcor Should we close this issue? I dont know if it is still on?
If it is, #494 could track that issue maybe

qarlosalberto, 2022-06-02:
> @std-max <https://github.com/std-max>, I'm not sure it's the same issue.
> @qarlosalberto <https://github.com/qarlosalberto> do you mind trying with
> a recent version of GHDL?

Sorry, I haven't tried it with a recent version.