VUnit/vunit #343: memory leak in the whisbone-master VC

TomGrossmann, 2018-06-14:
Hello,

I‘am currently using the whisbone-master VC from the Verification Component Library to transfer testdata to a moduleinterface. For a sufficient verification I have to send a lot of testdata via the Wishbone interface. To send much testdata, I repeat the call "write_bus(…)" in a loop (analogous to the loop in the "tb_wishbone_master.vhd" testbench). 

I have noticed the following: The simulation process allocating more and more memory in the course of the test and only releases it after the simulation has been completed. 
I did a test with the Wishbone testbench and its failing in the same way so there is obviously a memory leak here. I changed the following line in the file "vunit/vunit/vhdl/verification_components/run.py":

Line: 
`46    gen_wb_tests(test, [8, 32], [1, 64], [0.3, 1.0], [0.4, 0.0])`
Link: [https://github.com/VUnit/vunit/blob/52895e6f23167ab47773bbeb9b6e3a3a9e8edaf7/vunit/vhdl/verification_components/run.py#L46](url)
to:
`46    gen_wb_tests(test, [32], [2000000], [0.3], [0.4])`

When I start the wishbone-master testbench, the "wr block rd block" testcase is allocating more and more memory in the course of the test.
In the AXI testbench I could not see this behavior.

slaweksiluk, 2018-06-14:
I have not performed such a tests. As far as I remember I have some little local changes in my local repo, so I will try to reproduce it.

slaweksiluk, 2018-06-20:
I'm not too much experienced in debugging memory leaks problems. I use ghdl with gcc backend and I tried to use valgrind, however it reports variety of memory problems. Tracking memory leak is not easy as test bench executable is built from generated code.
@TomGrossmann How do you observe increasing memory usage? Simple system monitor memory usage history waveform is enough? I'm going to look for Vunit types usage patterns is examples sources and apply it in wishbone master vc, bit I need some reliable tool for memory leaks detection.

TomGrossmann, 2018-06-20:
I have observed the behavior in the standard Windows process analyzer. If I send a lot of test data via the Wishbone interface, the process caused a memory overflow.

The simulation process allocating more and more memory in the course of the test (up to 4GB). The process releases the memory after the simulation has been completed.

The following two images shows a snapshot of the Windows process analyzer during the simulation.

[image]
Figure 1: snapshot in the beginning of the simulation

[image]
Figure 2: snapshot just before the simulation has been completed

slaweksiluk, 2018-06-20:
Ok so during my tests with ghdl the problem occurred at allocate() function. It starts to fail after requesting 1MB of memory:
```vhdl
      memory := new_memory;
      buf := allocate(memory, 1024*1024);
```
I got segmentation fault from generated ghdl executable:
```
./tb_memory-a -grunner_cfg="active python runner : true,enabled_test_cases : Test 1MB alloc,output path : /home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/test_output/vunit_lib.tb_memory.Test_1MB_alloc_26b7b2a851fbe503d25ce6be39bce4f377471cd1/,tb path : /home/slawek/git/github/vunit/vunit/vhdl/verification_components/test/,use_color : true --assert-level=error"
Segmentation fault (core dumped)
```
strace:
```
/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/test_output/vunit_lib.tb_memory.Test_1MB_alloc_26b7b2a851fbe503d25ce6be39bce4f377471cd1/,tb path : /home/slawek/git/github/vunit/vunit/vhdl/verification_components/test/,use_color : true"
execve("./tb_memory-a", ["./tb_memory-a", "-grunner_cfg=active python runne"...], 0x7ffed4164f98 /* 58 vars */) = 0
brk(NULL)                               = 0x904000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=118157, ...}) = 0
mmap(NULL, 118157, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f7b7e05b000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\16\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=19264, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f7b7e059000
mmap(NULL, 2109680, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f7b7dc50000
mprotect(0x7f7b7dc53000, 2093056, PROT_NONE) = 0
mmap(0x7f7b7de52000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f7b7de52000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\340w\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1465608, ...}) = 0
mmap(NULL, 3449608, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f7b7d905000
mprotect(0x7f7b7da4f000, 2093056, PROT_NONE) = 0
mmap(0x7f7b7dc4e000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x149000) = 0x7f7b7dc4e000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\320#\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=94232, ...}) = 0
mmap(NULL, 2187272, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f7b7d6ee000
mprotect(0x7f7b7d704000, 2093056, PROT_NONE) = 0
mmap(0x7f7b7d903000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000) = 0x7f7b7d903000
mmap(0x7f7b7d904000, 8, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f7b7d904000
close(3)                                = 0
openat(AT_FDCWD, "/lib64/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0@\20\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=2066456, ...}) = 0
mmap(NULL, 3889792, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f7b7d338000
mprotect(0x7f7b7d4e4000, 2097152, PROT_NONE) = 0
mmap(0x7f7b7d6e4000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1ac000) = 0x7f7b7d6e4000
mmap(0x7f7b7d6ea000, 14976, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f7b7d6ea000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f7b7e057000
arch_prctl(ARCH_SET_FS, 0x7f7b7e057b80) = 0
mprotect(0x7f7b7d6e4000, 16384, PROT_READ) = 0
mprotect(0x7f7b7d903000, 4096, PROT_READ) = 0
mprotect(0x7f7b7dc4e000, 4096, PROT_READ) = 0
mprotect(0x7f7b7de52000, 4096, PROT_READ) = 0
mprotect(0x8b1000, 4096, PROT_READ)     = 0
mprotect(0x7f7b7e078000, 4096, PROT_READ) = 0
munmap(0x7f7b7e05b000, 118157)          = 0
brk(NULL)                               = 0x904000
brk(0x925000)                           = 0x925000
brk(NULL)                               = 0x925000
rt_sigaction(SIGSEGV, {sa_handler=0x5d09b1, sa_mask=[], sa_flags=SA_RESTORER|SA_RESETHAND|SA_SIGINFO, sa_restorer=0x7f7b7d36c6e0}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
mmap(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f7b7dfd6000
mmap(NULL, 528384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f7b7df55000
rt_sigaction(SIGSEGV, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f7b7d36c6e0}, NULL, 8) = 0
rt_sigaction(SIGSEGV, {sa_handler=0x5d09b1, sa_mask=[], sa_flags=SA_RESTORER|SA_RESETHAND|SA_SIGINFO, sa_restorer=0x7f7b7d36c6e0}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x7f7b7d36c6e0}, 8) = 0
openat(AT_FDCWD, "/home/slawek/git/github/vunit/vunit/vhdl/verification_components/vunit_out/test_output/vunit_lib.tb_memory.Test_1MB_alloc_26b7b2a851fbe503d25ce6be39bce4f377471cd1/vunit_results", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
fstat(3, {st_mode=S_IFREG|0664, st_size=0, ...}) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x7ffd32b9cb90} ---
--- SIGSEGV {si_signo=SIGSEGV, si_code=SI_KERNEL, si_addr=NULL} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)
```
@LarsAsplund @kraigher 
I tried to look into allocate() implementation, but at first time it's a mystery for me. What is maximum size of buffer I can request from allocate()?

LarsAsplund, 2018-06-20:
@slaweksiluk I started to look at the code and the first thing to notice is the following snippet in the testbench

``` vhdl
      for i in 0 to tb_cfg.num_cycles-1 loop
        write_bus(net, bus_handle, i*(sel'length),
            std_logic_vector(to_unsigned(i, dat_i'length)));
      end loop;
```

If `num_cycles` is 2000000 we will create and send that meny messages to the Wishbone master inbox. The message passing is asynchronous so this takes no simulation time which means that the master hasn't had time to consume these messages and there will be two million messages allocated.

What we need is a way to synchronize the testbench with the execution of the master. The first thing I would do is to add support for handling `wait_until_idle`. Have a look at http://vunit.github.io/com/user_guide.html#message-handlers-and-verification-component-interfaces-vci

Once you have that in place you can call `wait_until_idle` every 1000 write or so to limit the number of memory allocated  for messages.

LarsAsplund, 2018-06-20:
@slaweksiluk Another thing you can do is that you specify the actor in the `new_bus` call to have a limited inbox:

``` vhdl
  constant bus_handle : bus_master_t := new_bus(data_length => tb_cfg.dat_width,
      address_length => tb_cfg.adr_width, logger => master_logger, actor => new_actor(inbox_size => 1000));
```
In that way the `write_bus` call will start blocking when reaching that amount of unprocessed write commands. This is a start. It still seems a bit leaky. I can continue to look a bit later.

slaweksiluk, 2018-06-20:
Ok I will try some of this ideas. Currently I need to have ghdl segmentation fault fixed [ghdl issue611](https://github.com/ghdl/ghdl/issues/611).

LarsAsplund, 2018-06-21:
@TomGrossmann @slaweksiluk I need to look more into this but I found one leak in the communication package. Remove the initialization of the variable [here](https://github.com/VUnit/vunit/blob/master/vunit/vhdl/com/src/com_types.vhd#L524) rerun and see what happens.

TomGrossmann, 2018-06-21:
@LarsAsplund, thank you so much for the very fast answer. 

I have removed the initialization of the variables and the test runs excellent. 
You mentioned other leaks up there. 
> It still seems a bit leaky.

I do not know if the initialization was the whole problem, but the test does not allocate more much memory. It also runs a bit faster than before.

LarsAsplund, 2018-06-21:
@TomGrossmann @slaweksiluk I just released 3.6.2 which has the aforementioned bug fix so I'm closing this one.

closed 2018-06-21 by LarsAsplund