VUnit/vunit #345: Feature: Add a hook for post_report phase

krishnan-gopal, 2018-06-20:
Once an entire regression is finished, its very practical if the user could specify an action which can be done after that, for example sending an automated mail/sms to the user. This is particularly useful in case of long regressions which are submitted to a batch queue.
In the current implementation, everything that comes after the ui.main() is not executed, since the main function calls the end of the script automatically.
Therefore, a hook or an alternative could be used to specify any action that would take place after the summary/report generation is finished.

1138-4EB, 2018-06-20:
Did you check https://vunit.github.io/python_interface.html#pre-and-post-simulation-hooks ?

krishnan-gopal, 2018-06-20:
Yes. But the post_check() function is called for every testcase.
It would be good to have a hook which is called when the entire Vunit run is over, i.e. all the testcases are finished completely.

kraigher, 2018-06-20:
There is a post run hook for main http://vunit.github.io/python_interface.html#vunit.ui.VUnit.main

krishnan-gopal, 2018-06-21:
@kraigher 

I tried that, but the post_run function is being called before the start of compilation. Does this need to be fixed ?

kraigher, 2018-06-21:
@Krishnan you pass the result of calling the function rather than pass the function itself.  function() vs function

krishnan-gopal, 2018-06-21:
@kraigher 
Thats what I did. This is my code:

def inform_me():
    send_cmd = "sendmsg -s 'Simulation finished' \"myemail@email.com\" < /dev/null"
    print (send_cmd)
    os.system(send_cmd)
    
ui.main(post_run=inform_me())

kraigher, 2018-06-21:
Yes I suspected that was what you were doing and it is not correct. You need to pass the function as an argument and not the result of calling your function like you are doing now. Thus inform_me() should

krishnan-gopal, 2018-06-25:
I tried that as well, but I am getting the same result. The function is being called before compilation starts

krishnan-gopal, 2018-06-25:
Yes, that works now. Thanks.

closed 2018-06-25 by LarsAsplund