VUnit/vunit #346: Error with entity tb name

qarlosalberto, 2018-06-21:
Hello,

I have an error with the entity name when I use capital letters:

```
cruiz@DASPC034:~/repositorio/98410021_thresholdFilter/tb(1-creacion-de-core)$ python run.py --ide ise --clean
Traceback (most recent call last):
  File "run.py", line 170, in <module>
    tb_generated = thresholdFilter_top_tb_lib.entity("thresholdFilter_top_tb")
  File "/usr/local/lib/python2.7/dist-packages/vunit/ui.py", line 1229, in entity
    raise KeyError(name)
KeyError: 'thresholdFilter_top_tb'
```

If I use:

```
tb_generated = thresholdFilter_top_tb_lib.entity("thresholdfilter_top_tb")
```
I have not this error.

joshrsmith, 2018-06-22:
I am pretty sure that VUnit internally lowercases all found entity names. So this is expected behavior.

kraigher, 2018-06-25:
Yes VUnit lowercases entity names to handle the VHDL case insensitivity. Maybe it would be better to treat VHDL as case sensitive instead.

commit 3434dda7 (2018-06-29): Allow case-insensitive lookup of entities. Closes #346
closed 2018-06-29 by kraigher