VUnit/vunit #347: Interface to query Vunit files

davidmedinasigasi, 2018-06-25:
I propose  to add a '--vunitFiles' to list all the libraries and files that an external tool like Sigasi needs to import this project. 

The difference with '-f' parameter is that will only list the files that are built into vunit.

This is one of the steps for the Vunit integration with Sigasi Studio.

kraigher, 2018-06-25:
I would prefer if such an interface used the Python API. The command line interface is for end user usability not for tool integration. 

For example:
```python
from vunit import get_all_builtins
for library_name, file_name in get_all_builtins():
    print(library_name, file_name)
```

lievenlemiengre, 2018-06-25:
We don't want all builtin libraries, we only want to find libraries that are actually used.

What files are included and what libraries are available depends on the used VHDL version (and perhaps there may be more options in the future?).

Both are configured in run.py.

How about this: we could run a script to get the list of builtin library names. Using that list we could do "py run.py -f" and filter using the known builtin libraries.

kraigher, 2018-06-25:
Is your use model that the user maintains a `run.py` file that Sigasi does not touch?

lievenlemiengre, 2018-06-25:
Yes

kraigher, 2018-06-25:
In this case would it not be better to add a general `--export-to-json` flag that exported in JSON format all the information that a higher level tools would want to know?

By the way what is the need to differentiate between built-in files and user added files? Is it not between Sigasi-managed and VUnit-managed files that you want to differentiate? If you provided an project.csv to VUnit like you suggest in #348 then you can subtract those files from the ones reported by `--files` to get the set of built-in or non-"Sigasi-managed" files.

FranzForstmayr, 2018-06-29:
What's about a python method, which exports the vhdl files in a specific format. For example i would like a .tcl Script, which i can load in vivado, so that all source files are added to my project. Same method could be used for other vendor tools.

kraigher, 2018-08-24:
After further reflection I do not think using `--list` and `--files` for communicating with Sigasi is a good idea for the future and would rather see a `--export-to-json` flag to export data from VUnit to other tools. This means I do not think we should merge #365.

My reasons are as follows:
1. The `--list` and `--files` are mutually exclusive which means you need two invocations of `run.py` to get both. For large projects the `run.py` can be a bit sluggish especially if users add a lot of data-generation.

2. The `--list` and `--files` flags are Machine-to-User and not Machine-to-Machine and can change based on estetics etc. Maybe we would like to add color or extra information in the future. This would break the Sigasi integration.

Based on this reasons I think we should define a Machine-to-Machine data format for `--export-to-json` that conveys all information that Sigasi needs as well as a version of the format if we need to change it in the future.

Sorry to make this more complicated for you but I think we will be thankful in the future.

kraigher, 2018-08-25:
Regarding the format of the JSON export I think it should be something like this:
```json
{
    "version": "3.7.1",

    "files": [
        {
            "name": "tb_example.vhd",
            "library_name": "lib"
        },
        {
            "name": "tb_example_many.vhd",
            "library_name": "lib"
        }
    ],

    "test_cases": [
        {
            "name":  "lib.tb_example.all",
            "file_name": "tb_example.vhd",
            "lineno": 18,
            "multi_test_simulation": false
        },
        {
            "name":  "lib.tb_example_many.test_pass",
            "file_name": "tb_example_many.vhd",
            "lineno": 22,
            "multi_test_simulation": false
        },
        {
            "name":  "lib.tb_example_many.test_fail",
            "file_name": "tb_example_many.vhd",
            "lineno": 25,
            "multi_test_simulation": false
        },
    ]
}
```
The version would be the version of VUnit.
The names of the test cases corresponds to what you provide on the command line.
In case you are wondering the "multi_test_simulation" boolean would indicate that the test was part of a test suite where all tests run in the same simulation.

Using this format we can hide a lot of complexity in how VUnit creates tests case names based on configurations and such.

kraigher, 2018-08-25:
I can also think of another reason why using `--files` and `--list` is not a good idea. The user may add `print` statements to their run.py file and that output will also end up in stdout along with the files and list output.

kraigher, 2018-08-26:
I will have an implementation of `--export-json` ready by today so you do not have do implement it yourself.

kraigher, 2018-08-26:
I have implemented the `--export-json` command line argument on the `export-to-json` branch (https://github.com/VUnit/vunit/tree/export-to-json). You can read about the feature in the non-rendered documentation on the branch. https://raw.githubusercontent.com/VUnit/vunit/export-to-json/docs/cli.rst

1138-4EB, 2018-08-26:
@kraigher, since all the files must belong to one an only one library, would it be sensible to use this format instead?

``` json
 "files": {
  "lib": [
    "tb_example_many.vhd",
    "tb_example.vhd"
  ]
}
```

kraigher, 2018-08-26:
@1138-4EB I see the files and tests as lists of objects with fields. In your files proposal I do not see a nice way of adding more information about each file object without breaking backwards compatibility of the format or adding some other information on the side. With the format I implemented new fields can be added without breaking backwards compatibility.

1138-4EB, 2018-08-26:
@kraigher the main idea was to use library names to group objects. I now edited it according to your comment, in order to allow the addition of more fields without breaking backwards compatibility.

kraigher, 2018-08-26:
After your edit of the files proposal you have removed the disadvantage I mentioned. I cannot think of any objective difference between your current proposal and what I implemented. It is a matter of preference if you want to group files by libraries or view the library as a property of the file. 

Regarding your proposal to break out the library name from the test cases I would prefer to only use full test names in the export to avoid having the IDE needing to duplicate the VUnit logic for creating full test names from the fragments which would inhibit VUnit from changing how names are created in the future.

1138-4EB, 2018-08-26:
> I cannot think of any objective difference between your current proposal and what I implemented. It is a matter of preference if you want to group files by libraries or view the library as a property of the file.

My use case is to generate a list of files to be used in a TCL script that runs a synthesis tool. E.g.

``` tcl
for {set i 0} {$i < [llength $libs]} {incr i} {
  eval "set alib $[lindex $libs $i]"
  foreach j $alib {
    add_input_file [lindex $dirs $i]/hdl/$j.vhd -format {VHDL_2008} -work [lindex $libs $i]
  }
}
```

In order to get all the files that belong to a library, with you current proposal the name must be parsed. So, all the list/array must be processed to get/discard the files corresponding to a library. E.g. to ignore those in library 'test' or 'sim'.

A different context where grouping file/test according to the libraries is useful, is to show a tree view in a GUI. If the user wants to see all of them without hierarchy, it is easier for a tool to append fields rather than split them. However, as you say, changing how names are created might be a breaking change.

Note that these suggested changes are related to:

- Using a JSON config file and a 'generic' `run.py`. See [`config.json`](https://github.com/1138-4EB/vunit/blob/json_cfg_run/examples/vhdl/array_axis_vcs/config.json) and [`cfg_run.py`](https://github.com/1138-4EB/vunit/blob/json_cfg_run/examples/vhdl/array_axis_vcs/cfg_run.py).
- Exporting `VUnit.get_source_files`, `VUnit.get_compile_order` and/or `VUnit.get_implementation_subset` to JSON. See #352. E.g.:

``` python
files_dict = dict()
for x in vu.get_compile_order():
   try:
      files_dict[x.library.name].append(x.name)
   except:
      files_dict[x.library.name]=[x.name]

with open('compile_files.json', 'w') as outfile:
    json.dump(files_dict, outfile, indent=2)
```

kraigher, 2018-09-05:
@davidmedinasigasi @lievenlemiengre Have any thoughts on my recent comments regarding `--export-json`?

davidmedinasigasi, 2018-09-06:
@kraigher, We think the `--export-json` is the best option and we would look into using that instead the console commands in a future release.
Because we want full integration with VUnit, I have some questions regarding the export-json: 
- Are you planning to add the VUnit version to the json?
- Do you think it's possible to list the configurations in the json file?

kraigher, 2018-10-03:
@davidmedinasigasi @lievenlemiengre I just pushed the export json stuff to the master branch now. I also changed the test case line number location into a character offset + length to make it more precise.

davidmedinasigasi, 2018-10-25:
@kraigher I'm currently updating the VUnit integration with Sigasi with the new export-json. So far everything is working as expected with the json, but I'm getting wrong offsets in Windows and I think it's because of the newlines.

kraigher, 2018-10-25:
You are probably correct. I will investigate.

kraigher, 2018-10-25:
@davidmedinasigasi I tried here with files containing CRLF (\r\n) and I could not reproduce the problem. Sure the problem is not on your side?

kraigher, 2018-11-06:
@tivervac I could reproduce the problem and pushed a fix to master.

davidmedinasigasi, 2018-11-07:
@kraigher Thanks! When can we expect to see this in a release?

kraigher, 2018-11-07:
@davidmedinasigasi If you can confirm that it works I will make a release now.

davidmedinasigasi, 2018-11-09:
@kraigher It works perfectly

kraigher, 2018-11-09:
Good it is part of the latest release.

davidmedinasigasi, 2018-11-28:
@kraigher, I just noticed that when VUnit generates a Json for a system verilog test, the macro offsets are included in the test offset, is that intended?

kraigher, 2018-11-28:
What do you mean exactly?

davidmedinasigasi, 2018-11-28:
Does VUnit take in account the macro to calculate the test offset?

kraigher, 2018-11-28:
I am still not sure what you mean. The test case scanning if I recall correctly does just find the `TEST_CASE` string with a regex and the lexical position will be the string which is the test case name not including the `TEST_CASE` macro token. VUnit will not have done any verilog preprocessing to find the test case.

LarsAsplund, 2019-12-23:
The Sigasi integration is mature enough that this issue can be closed.

closed 2019-12-23 by LarsAsplund