VUnit/vunit #348: Load and map files with its respectives libraries from a csv description

davidmedinasigasi, 2018-06-25:
Maybe the first version of the Vunit integration with Sigasi will use this feature but we may change it in the future.

kraigher, 2018-06-25:
Why is this necessary. VUnit already has a Python-API to add files and libraries? Any user can add a .csv reader on top of this if they need to have one.

lievenlemiengre, 2018-06-25:
This api takes away a sore point to achieve integration with tools: managing the list of files provided by the user. If you think there is an easy way to get the same result, feel free to share your ideas. If you want good tool support there will be some inevitable impact on your api.

The only alternative that I see is a separate module (for example vunit-sigasi) with the csv parser. The user would have to install another module, add an import to their vunit file & importing the csv file would be slightly uglier. We can do it but I think this will give the end-user a better experience while the impact on your api is minimal. We're willing to contribute the code & tests.

kraigher, 2018-06-25:
I do not mind merging support for reading a list of files from CSV. I just ask why it is necessary for your use case to understand what your big picture is. Based on your answer I infer that you want this feature to be used by the end user whereas I first thought it was just about privately exchanging files between tools for which the Python API could be used without any additions.

qarlosalberto, 2018-06-28:
It would be interesting if Sigasi open sourced the code that interacts with VUnit :)

lievenlemiengre, 2018-07-02:
> It would be interesting if Sigasi open sourced the code that interacts with VUnit :)

We thought about this but it looks like we'll depend on a lot of internal API that may break on every Sigasi release. It's still likely that we'll open source parts of it. The com.sigasi.hdt.epl and com.sigasi.hdt.epl.ui plugins are shipped with source but they don't have a public repo.

kraigher, 2018-09-01:
This is already implemented.

closed 2018-09-01 by kraigher