VUnit/vunit #352: Python method to export dependencies for a given source for a specific IDE

FranzForstmayr, 2018-07-06:
I would like to have a Python method, which exports all dependencies for a given source file into a format, which common IDE's accept. 

I'm using vivado, which can read a .tcl file. So the method should export a tcl file, which i can read in vivado.
The method should list the files directly and libraries.
I can serve a pull request for this issue (only vivado), are there any additional functions, which we can provide easily?
Are there any doubts?

kraigher, 2018-07-06:
Have you looked at:
http://vunit.github.io/python_interface.html#vunit.ui.VUnit.get_source_files
http://vunit.github.io/python_interface.html#vunit.ui.VUnit.get_compile_order
http://vunit.github.io/python_interface.html#vunit.ui.VUnit.get_implementation_subset

FranzForstmayr, 2018-07-06:
I've looked at the first two, but not the latter one. This would be what i need, but i would like to have an IDE specific wrapper around, which is able to format the files properly. Do you want something like this in Vunit?

kraigher, 2018-07-06:
The IDE specific wrapper would be written by the user. I do not think it belongs in VUnit as the goal of VUnit is not to be a build tool that is used for synthesis. Ideally there is a higher level HDL package/module system that uses VUnit to run tests and something else to run synthesis.

FranzForstmayr, 2018-07-06:
I just tried the 'get_implementation_subset()'. I'm using the PoC library, but the lib is not visible in the listed implementation subset. I didn't set the 'no_parse' flag. Should the vhdl parser be able to parse the library dependencies or do i have to set the dependency manually?

1138-4EB, 2018-07-06:
Coming form [#347](https://github.com/VUnit/vunit/issues/347#issuecomment-401404084).

@FranzForstmayr I need a similar feature, also in TCL, but targeting Mentor Graphics' Precision RTL (Plus). I am considering to just export the implementation subset to a JSON file and then write a generic snippet composed of two nested for loops in TCL. Would you be interested in sharing this approach?

FranzForstmayr, 2018-07-07:
Hi, 
i wanted to try a python method, which exports a .tcl to get read in vivado. So no external tcl script should be necessary. I'll try this on my local fork the next days

FranzForstmayr, 2018-07-11:
@1138-4EB 
Ok, lets make a implementation subset in JSON.
I though about:
{
    "name"  : "myproject",
    "vhdl_standard" : "2008",
    "compile_options" : {
       "ghdl.flags" : ["flag1", "flag2"],
       "ghdl.elab_flags" : ["flag3", "flag4"]},
    "sim_options" : {
       "ghdl.flags" : ["flag1", "flag2"],
       "ghdl.elab_flags" : ["flag3", "flag4"]},
    "libraries" : {
        "vhdl_standard" : "specify, or use global defined above",
        "files" : [
            "src/Test1.vhd", "src/Test2.vhd", "tb/Test3_tb.vhd"
        ]
    }
}
Or should we split up the files, as you can define a different vhdl standard for each file. I don't know, if there's a simulator which supports different standards within one lib.

1138-4EB, 2018-08-26:
Sorry @FranzForstmayr, I missed your last message. Please, have a look at https://github.com/VUnit/vunit/issues/347#issuecomment-416055643 and https://github.com/VUnit/vunit/issues/347#issuecomment-416059855. As commented there, I think it is interesting to group files per library.

I am neither aware of how mixed standard versions are supported. So, expecifying it at the top level is ok for me.