VUnit/vunit #353: Combine multiple Vunit projects

FranzForstmayr, 2018-07-06:
I've read this post on SO:
https://stackoverflow.com/questions/42980036/how-to-combine-multiple-vunit-run-py-files-into-a-single-vunit-run

I want to start a discussion, if it is possible to provide a Pyhton API to import another vunit file, without changing old project's "run.py". Has anyone tried this before? Are there any risks?

If you start a "run.py" from a top project, all imported files should be loaded into a sub project. I would like to have the opportunity to compile the project into a specific library. The test runner will start the underlying tests, after that, the tests of the top module are executed.

kraigher, 2018-07-06:
In principle a run.py file cannot be imported, it is not a data file but rather an executable script. The data of the sub project run.py has to be factored out into a separate file which can be imported into the parent run.py file. In principle what you are asking for is a "module" system which VUnit does not implement. Such a module system should really not be part of VUnit but a separate system on top of VUnit since it should also handle synthesis and other parts of the HDL flow as well. A module system that only handles one aspect of the flow such as simulation is not really useful in my opinion. This is why I do  not want VUnit to have a module system and rather have it as a separate project. Internally at my company we have such a module system which uses VUnit underneath to run tests. 

Examples of open source module systems are fusesoc which has been integrated with VUint to some degree.

FranzForstmayr, 2018-07-06:
I'm aware about the script/import issue. I thought about some deeper python tricks, like parse the file with ast.

 I want to test a bigger signal chain in vivado. Let's assume you have a input signal, make some operations in module1, maybe a fft in module2 and finally format the data in a last module3. The top module consists of a testbench, which is instantiating the underlying modules. The underlying modules are vunit projects at it's own. 
This is of course possible in Vunit, when i manually include the source files into my top project. I just wanted the possibility to do this easily, i don't want to change existing structures of Vunit. 
 
Are you using Fusesoc? Or do you have other preferences?

kraigher, 2018-07-06:
I am not using fusesoc but have a comapny internal module system.

1138-4EB, 2018-07-06:
@FranzForstmayr, I think that you might find interesting the conversation that @dstadelm started on Jun 04 15:46 in [Gitter](https://gitter.im/VUnit/vunit). It is related to this SO question: https://stackoverflow.com/questions/42980036/how-to-combine-multiple-vunit-run-py-files-into-a-single-vunit-run

closed 2018-08-16 by kraigher